facebookarchive/Surround360 #135: How to remove 2448*2048 picture's pole?

dangsiper, 2016-11-11:
Hi,

We try to remove our picture's pole, but it seems doesn't work. Our picture's dimension is 2448*2048.
Below is our pic don't remove pole:

[image]
We used files in "res/pole_masks" to remove pole. Below is our pic remove pole:

[image]

Below is our original pic:
[image]
[image]
Is there any way to resolve this?
Thanks in advance.

aparrapo, 2016-11-11:
You have to create a mask around the pole and put those files in "res/pole_masks". See the example below:

[image]
[image]

dangsiper, 2016-11-12:
@aparrapo Thanks so much for your help.
We can remove the pole now.

closed 2016-11-12 by dangsiper