facebookarchive/Surround360 #136: The issue of the install control2

BroadDong, 2016-11-16:
[image]
Why？How could I do?

tellyarz, 2016-11-16:
Is your OS version 14? We had similar problem too until we switched to OS version 16. Seems libgtkmm-3.0-1v5 requires packages not found in version 14 but in version 16

BroadDong, 2016-11-16:
My system is Ubuntu14.04

emmanix2002, 2016-11-19:
@BroadDong you need to have Ubuntu 16.04 to build CameraControl2. The packages that gave you issues are available on >= 16.04. We ended up upgrading to 16.04 so we could setup CameraControl2.

BroadDong, 2016-11-20:
@emmanix2002 ，thank you very much！I have update it.But I encounter the problem，as follows.Could
you tell me how to solve it? I can't find the "CameraIspGen8.h".
[image]

emmanix2002, 2016-11-20:
@BroadDong the solution to that is as follow: 

- you'll need to setup `Halide` on your machine; you should build it yourself -- don't use the precompiled binaries; you'll run into an issue with that. This is covered in the setup instructions for the `surround360_render`.    
- Build `surround360_render` again, this time providing the `DHALIDE_DIR` which should point to the `cmake_build` inside the Halide directory - assuming you built Halide using `cmake`

After you've done the above, the required files will be created.

aparrapo, 2017-02-19:
Thanks @emmanix2002 for your input. Closing this for now. @BroadDong please reopen and comment if you still have problems.

closed 2017-02-19 by aparrapo