facebookarchive/Surround360 #139: purple around lamp after the isp processing

linming1247576468, 2016-11-23:
Did anyone meet the following issue?
[image]
[raw_data.zip](https://github.com/facebook/Surround360/files/608478/raw_data.zip)
note: Since github doesn't support tiff, so I put the raw image into zip file.

I captured the picture with my own camera, and ran the Raw2Rgb from Surround360 isp module. But there exist some purple around the lamps in ceiling. For a more accurate describing for pixel color ,I run the program with bilinear demosaicing. And the camera isp setting args are listed as follow.
"CameraIsp" : {
        "serial" : 0,
        "name" : "my camera",
        "bitsPerPixel" : 12,
        "width" : 1952,
        "height" : 1218,
        "compandingLut" :  [[0.0,  0.0, 0.0],
                            [0.5,  0.5, 0.5],
                            [1.0,  1.0, 0.0]],
        "blackLevel" : [5.0, 5.0, 5.0],
        "vignetteRollOff" : [[1.0, 1.0, 1.0],
                             [1.0, 1.0, 1.0],
                             [1.0, 1.0, 1.0],
                             [1.0, 1.0, 1.0]],
        "whiteBalanceGain" : [1.4509, 1, 1.6244],
        "stuckPixelThreshold" : 5,
        "stuckPixelDarknessThreshold" : 0.11,
        "stuckPixelRadius" : 0,
        "denoise" : 0.6,
        "denoiseRadius" : 2,
        "ccm" : [[1.780370, -0.523295, -0.258075],
                 [-0.706706,  1.994845, -0.288210],
                 [-0.013212, -0.600956,  1.581960]],
        "sharpenning" : [0, 0, 0],
        "saturation" : 1.1,
        "contrast" : 1.0,
        "lowKeyBoost" : [-0.2, -0.2, -0.2],
        "highKeyBoost" : [0.2, 0.2, 0.2],
        "gamma" : [0.4545, 0.4545, 0.4545],
        "bayerPattern" : "RGGB"
    }

I had changed the blackLevel, whiteBalanceGain and other args in different values, but the purple couldn't be removed anyway. Does anyone know about the reason? Thanks.

linming1247576468, 2016-11-23:
From some materials, I learned about chromatic aberration which may cause the problem. But I didn't see any code about fixing this issue in Surround360. Does the point gray camera dealing well with this problem?

TomHVR, 2016-11-25:
https://en.wikipedia.org/wiki/Purple_fringing

It's an optical property of the lens. Best you can do is take it to an external program like Davinci Resolve, chromakey the purple color and de-saturate it.

closed 2017-02-11 by linming1247576468