facebookarchive/Surround360 #142: Can't record from the  CameraControlUI?

BroadDong, 2016-11-24:
when I get the CameraControlUI，but I can't record.And I only gain the one picture from the first camera.Please tell me Why? Thank you very much. 
[image]

emmanix2002, 2016-11-25:
@BroadDong that is just the preview window. It shows you the "view" from the selected camera; since you have all 17 cameras connected, recording takes data from all 17 of them and writes it to your output directory. You can confirm this by looking at the output after using `surround360_render`.

aparrapo, 2017-02-19:
@BroadDong, closing this issue for now, but let us know if the problem persists :)

closed 2017-02-19 by aparrapo

lcyzzz, 2017-04-21:
Hi @BroadDong, I have the same problem with yours and I  want to know did you fix this problem?