facebookarchive/Surround360 #149: Sample Video Unpack issues

kientran, 2016-12-04:
Having some trouble unpacking the sample video.

Pulled the latest version of the render code and compiled sucesfully.  Attempting to unpack the sample bin files but am getting path errors.

EditBay-01% python /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/scripts/run_all.py --rectify_file NONE --quality 6k --cubemap_height 0 --cam_to_isp_config_file /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/config/isp/cam_to_isp_config.json --nbits 8 --data_dir /Volumes/8TB-Stripe/Kien/BL360/fb360/palace3 --rig_json_file /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/config/17cmosis_default.json --image_width 2048 --dest_dir /Volumes/8TB-Stripe/Kien/BL360/fb360/export --steps_unpack  --flow_alg pixflow_low --cubemap_format video --src_intrinsic_param_file /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/config/sunex_intrinsic.xml --start_frame 0 --pole_masks_dir /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/pole_masks --cubemap_width 0 --image_height 2048 --frame_count 0

-------- Sat Dec 03 2016 19:42:27 CST -------

** UNPACK ** [Step 1 of 1]

E1203 19:42:27.795583 1942266624 SystemUtil.cpp:48] Terminated with VrCamException: error opening binary file. err: No such file or directory filename: /Volumes/8TB-Stripe/Kien/BL360/fb360/palace3/palace3_/0.bin

Any thoughts why it seems like it's adding in an extra path slash?

I am attempting to run this on OSX 10.10.5 with python 2.7.11 from homebrew.

mateuszb, 2016-12-14:
Paths passed in is arguments should be passed with a trailing slash, e.g. /Volumes/8TB-Stripe/Kien/BL360/fb360/palace3/

kientran, 2016-12-26:
THanks for the suggestion, but it sitll has no effect.  Based on the output logs, it looks like it's getting confused by the "_"

python /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/scripts/run_all.py --rectify_file NONE --quality 6k --cubemap_height 0 --cam_to_isp_config_file /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/config/isp/cam_to_isp_config.json --nbits 8 --data_dir /Volumes/8TB-Stripe/Kien/BL360/fb360/palace3/ --rig_json_file /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/config/17cmosis_default.json --image_width 2048 --dest_dir /Volumes/8TB-Stripe/Kien/BL360/fb360/export/ --steps_unpack  --flow_alg pixflow_low --cubemap_format video --src_intrinsic_param_file /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/config/sunex_intrinsic.xml --start_frame 0 --pole_masks_dir /Volumes/8TB-Stripe/Kien/BL360/fb360/Surround360/surround360_render/res/pole_masks --cubemap_width 0 --image_height 2048 --frame_count 0

** UNPACK ** [Step 1 of 1]

E1226 00:53:55.178452 2069005056 SystemUtil.cpp:48] Terminated with VrCamException: error opening binary file. err: No such file or directory filename: /Volumes/8TB-Stripe/Kien/BL360/fb360/palace3//palace3_/0.bin

aparrapo, 2017-02-19:
@kientran, I believe the solution in #158 should fix your problem. Closing this for now, but feel free to reopen it if the problem persists :)

closed 2017-02-19 by aparrapo