facebookarchive/Surround360 #153: Camera Control UI does not Start

Realit7, 2016-12-12:
[image]

Hi all,
When I run the CameraControlUI this message appears. I am unable to start the program. Has anyone experienced the same issue?

The first issue occurred when I built ffmpeg. I ended up taking the way described in issue https://github.com/facebook/Surround360/issues/83. After that I installed the capture UI and it installed. However after running the software it seems there was a serious error while compiling the software as the above shown error message came up.

I desperately need some help. You can also contact me under info@reality7.com.

Thank you for your help in advance.

Max

mateuszb, 2016-12-14:
what version of the source code are you using?

malcolmbrown, 2016-12-15:
Hi. Do you have the raid mounted at /media/snoraid ?

Reality7, 2016-12-15:
I am running the latest version of the source code. I downloaded it today and did a clean install.

Reality7, 2016-12-15:
Hi, I created a directory in /media called md0raid. I mounted the raid at
/media/md0raid. Is this wrong?

*Max Konrad*CEO

Tel: +49 40 368 800 763
Mobil: +49 151 419 66 263
Mail: max.konrad@reality7.com
Internet: www.reality7.com (forthcoming)

Impressum: Reality7 GmbH & Co. KG | Fuhlsbüttler Straße 29 | 22305 Hamburg
| Germany

Sitz Hamburg | AG Hamburg HRA 119284 | USt.-ID. Nr: DE301697102 | pers.
haftende Gesellschafterin:  ??? GmbH | Sitz Hamburg | AG Hamburg HRB 138071 |
Geschäftsführer: Max Konrad

---

Diese E-Mail ist nur für den Adressaten bestimmt und kann vertrauliche
und/oder rechtlich geschützte Informationen enthalten. Wenn Sie nicht der
richtige Adressat sind oder diese E-Mail irrtümlich erhalten haben,
informieren Sie bitte sofort den Absender und vernichten Sie diese Mail.
Das unerlaubte Kopieren sowie die unbefugte Weitergabe dieser Mail ist
nicht gestattet.

This e-mail may contain confidential and/or privileged information. If you
are not the intended recipient (or have received this e-mail in error)
please notify the sender immediately and destroy this e-mail. Any
unauthorised copying, disclosure or distribution of the material in this
e-mail is strictly forbidden.

Reality7, 2016-12-18:
I have an update on this problem. First of all I changed the hardware and did not use the camputer hardware named in the manual - mine is "better". It seems that the capture software was developed specifically for that camputer build though and that is the reason why so many people are having issues with building the camera - they are not strictly sticking to the plan. Facebook should update that in their manual - "please use this camputer build otherwise we cannot guarantee that the software works" should be a sufficient hint.

So to the problems:
1) I have 16 x Samsung 850 Pro 2 TB SSDs. Therefore, I have 2 x hardware RAIDs and a software Raid to combine them. This means my RAID is called /md0. If you build the FB camputer you have one RAID and you automatically create a RAID named sda in the directory dev (/dev/sda). This is coded into the capture software and has to be changed manually before installing - otherwise you get my error.

2) I used a SuperMicro motherboard with 6 x Pcie 3.0 x8 and 1 x Pcie 3.0 x16 solts. I attached two hardware Raids, four UE-1004 Cards (16 x USB 3.0 slots) and the GPU (I used the same FB used). I connected one Camera to one USB 3.0 on the mainboard and the other 16 x USB 3.0 cameras to the cards. This is also an issue as the capture software can only identify 16 cameras and, therefore, creates an error. It seems the software needs 5 x UE-1004 cards with the 17 cameras attached to the cards in order to work. Btw  flycap identifies all 17 cameras  so it must be the capture software.

3) We have 2 Xenon chips and FB camputer only has one. This is only a hunch but this also seems to be a conflict for the software. I will confirm though as soon as I found out.

4) We are using server RAM and for some reason the software is unable to find the memory. This is also a hunch but we are working on it to be certain.

So these are only a few problems that occurred while building the software. We will keep you informed as soon as we have solved the problem. Additionally we will write an install script including all the dependencies such as ffmpeg - which has caused many problems for other people as well.

Reality7, 2017-01-18:
In the end this issue unfortunately boiled down to the fact that the included "supereasyjson" lib doesn't cater for locales with a different decimal separator (e.g. comma in German) which will lead to floating point values read from the json being read as NULL values and thereby causing the above mentioned error of an unexpected json structure - since NULL obviously is not an ObjectVal ...

The quick workaround for us has been to switch the system's locale to en_US, but we're hoping to supply a pull request for this soon.

aparrapo, 2017-02-19:
Nice catch @Reality7! Closing this issue, but feel free to submit a pull request if you ever get to fix this :)

closed 2017-02-19 by aparrapo