facebookarchive/Surround360 #155: confusion about calculating the overlapAngleDegrees

linming1247576468, 2016-12-26:
In the method 'generateRingOfNovelViewsAndRenderStereoSpherical' of TestRenderStereoPanorama.cpp, there are codes calculating the overlap infos of cameras: **const float overlapAngleDegrees = (camFovHorizontalDegrees * float(numCams) - 360.0) / float(numCams);** But my understanding here is that only when all cameras' optical centers are at the same position( center of the rig) can the formula works. But actually the cameras are placed along the circle with radius 'r', so how could the codes still work? Did I misunderstand the codes or miss something? Thanks if someone could explain it.

linming1247576468, 2016-12-26:
When considering the problem above, I met with another question: what will happen if I change the 'r' as argument even though the physical r is unchanged. When my cameras setting can't meet the requirement of r*sin(fov/2-360/n)>3.2, I increased the r of the json file, and the result was just good although the disparity may decrease.

fbriggs, 2017-01-03:
Q1: The formula for overlapAngleDegrees is only correct if all cameras are at the same position?
A1: You are right that the formula for overlapAngleDegrees assumes all of the cameras are at the same position (0, 0, 0). Most of the steps of the rendering pipeline are done in equirectangular projection. Near the beginning, each camera's original image is projected to equirectangular. In this projection, it is assumed everything is at a distance of infinity. This is geometrically equivalent to putting all of the cameras in the same spot (the radius r is much smaller than infinity). There are advantages to doing this, e.g., everything that really is at a distance near infinity will have zero optical flow, and optical flow starts with an initial guess of 0, so its already right. 

Q2: What happens if your input for r' doesn't match the physical value of r?
A2: It can still produce an output. It will be equivalent to scaling the world by the ratio r'/r. Note that won't make any difference for objects that are very far away. It will introduce some scaling distortion for objects that are closer.

closed 2017-01-03 by fbriggs