facebookarchive/Surround360 #159: some links to help understand the projections ?

jiangjingping, 2017-01-05:
It seems that the projection part is very important in the render system, I have some trouble understanding it,  and  the link (https://www.facebook.com/pxlcld/nGpT)  you give in the comment is not available now,  can you give me another link or some papers to help ?
thanks a lot :)

fbriggs, 2017-01-05:
The projection we use is equirectangular, which is the same a spherical coordinates. In an equirect video, the x-axis corresponds to an angle theta, and the y-axis corresponds to an angle phi. Each pixel (x, y) corresponds to an angle (theta, phi). The spherical-to-cartesian transform (https://en.wikipedia.org/wiki/Spherical_coordinate_system) maps (theta, phi), to a (dx, dy, dz) which is a vector describing the ray direction for light through that pixel. For some math convenience reasons, we do these projections as though all of the cameras are at the origin, even though that isn't really the case (we account for this later). 

Here's Google's paper about JUMP. It uses the same projections at the output: https://research.google.com/pubs/pub45617.html

Paul Bourke has a lot of good notes about these projections:
http://paulbourke.net/geometry/transformationprojection/

closed 2017-01-05 by fbriggs

jiangjingping, 2017-01-06:
@fbriggs thank u so mach ~, now I have another question about the optical flow code.  in function combineNovelViews() ,  to get the final blend image , you use a formula `float(colorL[0]) * lerp(blendL, softmaxL, deghostCoef) + float(colorR[0]) * lerp(blendR, softmaxR, deghostCoef),  `  , I don't quiet understand  how this formula works and what the arguments mean , can you  explain a little or give me some reference ? , thank u :)

fbriggs, 2017-01-06:
@jiangjingping  - please open a new issue if you have a new question; it helps us with book keeping.

Every pixel in the output image is a weighted average of two pixels from two different adjacent cameras. We apply optical flow to each pair of adjacent cameras, and that gives a flow field that warps one camera's image to look like the other, and vice-versa. For each column of pixels in the final equirect output, it corresponds to a particular ray direction. If that ray happens to go right through a real camera, we can get a color without warping that image at all. We could also get the color for that ray by warping the other camera's image a lot. We prefer the pixel color we get from the less warped image. This is part of the motivation for the formula you posted.

So lets consider a simplified version of what we do: take the average of two pixels, and weight the average by how much you had to warp each image by optical flow to get the pixels. That way you will get the right color for rays that go right through real cameras. This simplified approach has some problems. For example, if there are any optical flow errors, on in cases like occlusions, they two pixel colors we get from flow/warping may be very different. If we just take a weighted average, the result will look "ghosting". So instead, we look at the two colors that are going to be blended, and if they are similar we blend them, but if they are too different, instead we want to pick just one or the other color. In this case, we pick the color which is (a) warped less, and (b) corresponds to a larger magnitude flow, which suggests it is closer. Part (b) here helps to produce a better answer in the case of occlusions. All of the "IF" statements in the description above aren't really hard yes/no values, but instead they are are coefficients between 0 and 1 that weight how much we are doing a simple linear blend, vs. a blend that perfers one or the other color. The logic of the IF statements is smoothed out using softmax functions.

jiangjingping, 2017-01-07:
@fbriggs ok I got it , that's really an ingenious idea !  thanks for your nice explanation :)

jiangjingping, 2017-01-07:
@fbriggs ok I got it , that's really an ingenious way !  thanks for your nice explanation :)