facebookarchive/Surround360 #160: problem of optical  flow around moving object

electech6, 2017-01-13:
Hi, I have some problem with the optical flow around the boundary of the moving object. The left result shows some undesirable "ghost", while the right result is quite normal. 
the left result  
[image]
the right result 
[image]
I have got some  knowledge of  combineNovelViews from [here](https://github.com/facebook/Surround360/issues/159). The softmax function are used to decide to do A alternative blend or a linear blend. Although this method is quite ingenious, it seems little effect on my case. could you give me some suggests ? Many thanks.

fbriggs, 2017-01-17:
This is a limitation of our current optical flow algorithm. In cases like this, only one camera sees part of the scene, and it doesn't have enough information to get the right answer for that part. There isn't much that can be done about it, other than:
1) Change the hardware of the camera. For optimal results, the distance between adjacent cameras should be minimized. Making the whole camera array smaller helps a lot with this.
2) Keep things further away from the camera. This has the same effect as making the camera smaller, relatively.
3) Improve our optical flow code. How to get the most accurate optical flow with a fast runtime is an open research problem in computer vision. Currently our implementation compares very well with other optical flow algorithms that are published in literature, so making progress in this area is hard. However, it is not impossible, so if you have resources to spend there, it may be worth while.

closed 2017-01-17 by fbriggs

jiangjingping, 2017-05-19:
@fbriggs Can you give me some papers(or other reference) about your optical flow algorithm? I am interested  and want to learn more about your optical flow algorithm,Maybe to see if there is any chance to improve the code, perhaps~.thanks in advance :)