facebookarchive/Surround360 #161: confused about renderStereoPanoramaChunks

jiangjingping, 2017-01-13:
Hi~, I am very interested in the way you render stereo panorama ,but these days I faced some questions about that:
Q1: in function renderStereoPanoramaChunksThread()
     `lazyNovelViewBuffer.warpL[currChunkX][v] =
         Point3f(slabShift + vergeAtInfinitySlabDisplacement, v, shift);
      lazyNovelViewBuffer.warpR[currChunkX][v] =
        Point3f(slabShift - vergeAtInfinitySlabDisplacement, v, shift);`
what is 'slabShift' and 'vergeAtInfinitySlabDisplacement' ?
Q2: when we use lazyNovelViewBuffer to render the chunks , in function `renderLazyNovelView()`, we use the mat 'warpCompositon' to remap the src image:
      `warpOpticalFlow.at<Point2f>(y, x) = Point2f(lazyWarp.x, lazyWarp.y);`
      `warpComposition.at<Point2f>(y, x) = Point2f(lazyWarp.x + flowDir.x * t, lazyWarp.y + flowDir.y * t);`
why we should use 'warpComposition' to get the chunk ?
hope you could make me clear :)

fbriggs, 2017-01-17:
Q1: see https://github.com/facebook/Surround360/issues/116 and https://github.com/facebook/Surround360/issues/65

Q2: opencv's cv::remap function takes an image, and a "warp Mat" which specifies for each pixel in the output, which pixel of the input image to sample (possibly non-integer coordinate). We could apply that function twice, each time with a different warp Mat, but instead, we build a warpMat that has the same effect as applying two consecutive calls to cv::remap. That is what is stored in warpComposition. The purpose of the warpComposition is this: at any point between a pair of real cameras, we could make a virtual camera by warping the two images based on the optical flow between them. That would give us a full novel view images. That is the first remap in the composition. However, we only need one column of pixels from that novel view, and those pixels will end up in a chunk of the output panorama. Getting that column of pixels is the second remap in the composition.

closed 2017-01-17 by fbriggs

jiangjingping, 2017-01-19:
@fbriggs thank u ~