facebookarchive/Surround360 #164: build broken after on ubuntu 14.04 after commit 4573382

ybrs, 2017-02-01:
after the commit 45733828da4fbe4d2aabb171ccc908a34919519a 

cmake gives errors for boost,

```
$ ~/build/Surround360/surround360_render$ cmake -DCMAKE_BUILD_TYPE=Release
-- Found required Ceres dependency: Eigen version 3.2.0 in /usr/local/include
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.12.0 installed in: /usr/local with components: [LAPACK, SchurSpecializations, OpenMP]
CMake Warning at CMakeLists.txt:8 (FIND_PACKAGE):
  By not providing "Findboost.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "boost", but
  CMake did not find one.

  Could not find a package configuration file provided by "boost" with any of
  the following names:

    boostConfig.cmake
    boost-config.cmake

  Add the installation prefix of "boost" to CMAKE_PREFIX_PATH or set
  "boost_DIR" to a directory containing one of the above files.  If "boost"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/vagrant/build/Surround360/surround360_render
```

and if you fix that by changing `boost` to `Boost`, cmake complains about libdouble-conversion 

```
[ 50%] Building CXX object CMakeFiles/GeometricCalibration.dir/source/calibration/GeometricCalibration.cpp.o
/home/vagrant/build/Surround360/surround360_render/source/calibration/GeometricCalibration.cpp: In function ‘int main(int, char**)’:
/home/vagrant/build/Surround360/surround360_render/source/calibration/GeometricCalibration.cpp:884:55: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
   system(std::string("mkdir -p " + outputDir).c_str());
                                                       ^
/home/vagrant/build/Surround360/surround360_render/source/calibration/GeometricCalibration.cpp:889:56: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
     system(std::string("mkdir -p " + debugDir).c_str());
                                                        ^
Linking CXX executable bin/GeometricCalibration
/usr/bin/ld: CMakeFiles/GeometricCalibration.dir/source/calibration/GeometricCalibration.cpp.o: undefined reference to symbol '_ZNK17double_conversion23DoubleToStringConverter11ToPrecisionEdiPNS_13StringBuilderE'
/usr/local/lib/libdouble-conversion.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [bin/GeometricCalibration] Error 1
make[1]: *** [CMakeFiles/GeometricCalibration.dir/all] Error 2
make: *** [all] Error 2
```

adding `double-conversion` to TARGET_LINK_LIBRARIES in CMakeLists.txt fixes that too.

here is the total diff.

```
diff --git a/surround360_render/CMakeLists.txt b/surround360_render/CMakeLists.txt
index a2b7736..9442e5e 100644
--- a/surround360_render/CMakeLists.txt
+++ b/surround360_render/CMakeLists.txt
@@ -4,7 +4,8 @@ PROJECT(Surround360Render CXX)

 FIND_PACKAGE(OpenCV)
 FIND_PACKAGE(Ceres REQUIRED)
-FIND_PACKAGE(boost COMPONENTS filesystem system)
+
+FIND_PACKAGE(Boost COMPONENTS filesystem system)

 INCLUDE_DIRECTORIES(${OpenCV_INCLUDE_DIRS})
 INCLUDE_DIRECTORIES(${CMAKE_CURRENT_SOURCE_DIR}/source)
@@ -530,6 +531,7 @@ TARGET_LINK_LIBRARIES(
   folly
   boost_filesystem
   boost_system
+  double-conversion
   ${OpenCV_LIBS}
   ${PLATFORM_SPECIFIC_LIBS}
   ${CERES_LIBRARIES}
```

in case you might wanted to know. 

thank you.

aparrapo, 2017-02-01:
Hi @ybrs . Thanks for the fix! If you want you can make a pull request and we'll accept it, so we can give you proper recognition. Else just let me know and I'll push the diff myself :)

ybrs, 2017-02-01:
Hi @aparrapo thanks for fast response. I already had a branch so, just sent the PR :)

Thank you.

closed 2017-02-09 by aparrapo