facebookarchive/Surround360 #166: Issues while running "Geometric Calibration"

emmanix2002, 2017-02-09:
I've been running into issues while trying to complete the **geometric calibration** for our rig. Every time I run the command, I keep getting an exception at the point where the script is trying to convert the SQLite db data to JSON.    
```
==============================================================================
Exhaustive feature matching
==============================================================================

Matching block [1/2, 1/2] in 169.994s
Matching block [1/2, 2/2] in 0.373s
Matching block [2/2, 1/2] in 44.220s
Matching block [2/2, 2/2] in 2.183s
Elapsed time: 3.614 [minutes]

feature matching runtime: 0:03:37.336710

Converting database to JSON file...
Traceback (most recent call last):
  File "scripts/geometric_calibration.py", line 172, in <module>
    features_db_to_json(colmap_db_path, matches_json)
  File "scripts/geometric_calibration.py", line 87, in features_db_to_json
    inlier_matches = np.fromstring(row[1], dtype=np.uint32).reshape(-1, 2)
TypeError: argument 1 must be string or read-only buffer, not None
```
What you see above is the issue, I've browsed the SQLite db and it runs into this issue when using rows where the `data` column is NULL.
Any advice or solution for this?

emmanix2002, 2017-02-09:
I went past this issue by adjusting the `SQL` query in `geometric_calibration.py` on `line 83` from this:

```
cursor.execute("SELECT pair_id, data FROM matches;")
```

**to**    

```
cursor.execute("SELECT pair_id, data FROM matches WHERE data IS NOT NULL;")
```

closed 2017-02-09 by emmanix2002

aparrapo, 2017-02-09:
Hi @emmanix2002 , thanks for catching this bug! If you want you can make a pull request with this change so your contribution can be properly recognized :)

emmanix2002, 2017-02-09:
@aparrapo I'll do that as soon as I'm on my computer :-)

emmanix2002, 2017-02-10:
@aparrapo sent the PR