facebookarchive/Surround360 #167: Issues running geometric calibration

emmanix2002, 2017-02-09:
After going past the issue #166 I've run into another problem. Below is the console output:

```
==============================================================================
Exhaustive feature matching
==============================================================================

Matching block [1/2, 1/2] in 169.922s
Matching block [1/2, 2/2] in 0.379s
Matching block [2/2, 1/2] in 44.307s
Matching block [2/2, 2/2] in 2.195s
Elapsed time: 3.615 [minutes]

feature matching runtime: 0:03:37.321253

Converting database to JSON file...
Running geometric calibration...

 /home/fb/Software/Surround360/surround360_render/bin/GeometricCalibration --json \/home\/fb\/Software\/Surround360\/surround360\_render\/res\/config\/17cmosis\_rig\.json --output_json \/home\/fb\/Desktop\/Surround360\-Configs\/17cmosis\_calibrated\.json --matches \/home\/fb\/Desktop\/geometric\_calibration\/vid\/\/matches\.json --pass_count 10 --log_dir \/home\/fb\/Desktop\/geometric\_calibration\/vid\/\/logs --logbuflevel -1 --stderrthreshold 0  --debug_matches_overlap 0.2 --save_debug_images 

I0209 11:39:19.314270 21867 GeometricCalibration.cpp:149] 17 images loaded
I0209 11:39:19.316267 21867 GeometricCalibration.cpp:197] 16410 keypoint observations loaded
I0209 11:39:19.343530 21867 GeometricCalibration.cpp:927] RMSEs: Pos 0 Rot 2.51037e-16 Principal 0 Distortion 0 Focal 0 Angle 4.1528e-16 
I0209 11:39:27.440140 21867 GeometricCalibration.cpp:364] removeOutliers: total: 8205 invisible: 2897 outliers: 31
I0209 11:39:27.441366 21867 GeometricCalibration.cpp:453] found 2822 traces
I0209 11:39:29.232239 21867 GeometricCalibration.cpp:364] removeOutliers: total: 5277 invisible: 0 outliers: 27
I0209 11:39:29.233448 21867 GeometricCalibration.cpp:453] found 2823 traces
F0209 11:39:47.590806 21867 problem_impl.cc:568] Parameter block not found: 0x2ab9770. You must add the parameter block to the problem before it can be set constant.
*** Check failure stack trace: ***
    @     0x7f0957e0b5cd  google::LogMessage::Fail()
    @     0x7f0957e0d433  google::LogMessage::SendToLog()
    @     0x7f0957e0b15b  google::LogMessage::Flush()
    @     0x7f0957e0de1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x47a1e5  ceres::internal::ProblemImpl::SetParameterBlockConstant()
    @           0x462d4c  refine()
    @           0x4562f7  main
    @     0x7f0952a76830  __libc_start_main
    @           0x457199  _start
    @              (nil)  (unknown)
Aborted (core dumped)

calibration runtime: 0:00:28.809187
```

You can check the generated [matches.json](https://dl.dropboxusercontent.com/u/25236124/matches.json) file here if you need to. 
All suggestions are welcome.

tox-fb, 2017-02-10:
hey, i looked at the matches.json, and it looks like it matched the /raw/ images and the /flow_images/

the matching needs to happen between /isp_out/ images

please refer to the readme (https://github.com/facebook/Surround360/blob/master/surround360_render/CALIBRATION.md) for details, specifically
"Unpack the frames and run the ISP step to get RGB images. Put them in a separate directory. For this example we asumme they are in ~/Desktop/geometric_calibration/000000/isp_out/cam[0-16].png"

thank you for attaching the matching.json, this is how i figured it out:
> grep -A 4 "image1" matches.json
...
--
            "image1": "000000/flow_images/bottomImage2.png",
            "image2": "000000/raw/cam0.tiff",
            "matches": [
                {
                    "idx1": "7",
--
            "image1": "000000/flow_images/bottomImage2.png",
            "image2": "000000/raw/cam1.tiff",
            "matches": [
                {
                    "idx1": "133",
--
            "image1": "000000/flow_images/bottomImage2.png",
            "image2": "000000/raw/cam10.tiff",
            "matches": [
                {
                    "idx1": "116",
--
            "image1": "000000/flow_images/bottomImage2.png",
            "image2": "000000/raw/cam11.tiff",
            "matches": [
                {
                    "idx1": "7",
--

tox-fb, 2017-02-10:
i am going to go ahead and close this issue, if this does not solve the problem, please reopen!

closed 2017-02-10 by tox-fb

emmanix2002, 2017-02-10:
@tox-fb I'll look into it, and try it again on monday. I'll let you know how it goes, thanks for the help.

sdlcwangsong, 2018-05-20:
Hi, @emmanix2002 , did you solve this problem? how to solve it?