facebookarchive/Surround360 #169: unknown command line flag 'accelerate'

XQ1560, 2017-02-11:
Hi all,
i'm running Surround 360 - Process Dataset to test sample_dataset(2 frames). And then, i got a core dump error. Here are my error report and config.

-------- Sat Feb 11 2017 13:08:37 CST -------

python /home/dyq/Desktop/Surround360/surround360_render/scripts/run_all.py --rectify_file /home/dyq/Desktop/sample_dataset/rectify.yml --enable_top  --quality 6k --cubemap_height 0 --cam_to_isp_config_file /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/cam_to_isp_config.json --steps_isp  --steps_rectify  --nbits 8 --data_dir /home/dyq/Desktop/sample_dataset --rig_json_file /home/dyq/Desktop/Surround360/surround360_render/res/config/17cmosis_default.json --image_width 2048 --dest_dir /home/dyq/Desktop/sample_dataset --flow_alg pixflow_low --cubemap_format video --src_intrinsic_param_file /home/dyq/Desktop/Surround360/surround360_render/res/config/sunex_intrinsic.xml --start_frame 0 --save_debug_images  --steps_render  --enable_pole_removal  --verbose  --pole_masks_dir /home/dyq/Desktop/Surround360/surround360_render/res/pole_masks --cubemap_width 0 --enable_bottom  --enable_render_coloradjust  --image_height 2048 --frame_count 1

** ISP ** [Step 1 of 3]

 python /home/dyq/Desktop/Surround360/surround360_render/scripts/batch_process_isp.py --surround360_render_dir /home/dyq/Desktop/Surround360/surround360_render --root_dir /home/dyq/Desktop/sample_dataset --start_frame 0 --end_frame 0 --cam_to_isp_config_file /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/cam_to_isp_config.json --nbits 8  --verbose 

----------- [ISP] processing frame: 000000
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam3.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp3.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam3.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam7.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp7.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam7.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam0.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp0.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam0.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam4.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp4.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam4.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam12.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp12.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam12.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam8.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp8.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam8.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam9.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp9.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam9.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam10.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp10.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam10.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam1.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp1.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam1.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam5.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp5.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam5.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam16.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp16.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam16.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam6.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp6.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam6.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam15.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp15.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam15.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam14.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp14.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam14.png --output_bpp 8 --accelerate  
 /home/dyq/Desktop/Surround360/surround360_render/bin/Raw2Rgb --logbuflevel -1 --log_dir /home/dyq/Desktop/sample_dataset/logs --stderrthreshold 0 --input_image_path /home/dyq/Desktop/sample_dataset/vid/000000/raw/cam11.bmp --isp_config_path /home/dyq/Desktop/Surround360/surround360_render/res/config/isp/isp11.json --black_level_offset 0 --output_image_path /home/dyq/Desktop/sample_dataset/vid/000000/isp_out/cam11.png --output_bpp 8 --accelerate  
[... cut ...]
 0, 1266.978951119106, 1023.418452916802;
 0, 0, 1]
I0211 13:08:37.213238  9034 TestRingRectification.cpp:73] [-0.3203872316005091, 0.09807279936742816, 0.0002792073519266573, -0.000301939521590532, -0.01352514480488489]
I0211 13:08:37.213244  9034 TestRingRectification.cpp:75] reading camera json file
I0211 13:08:37.213493  9034 TestRingRectification.cpp:82] removing top and side cameras from rig

[image]
[image]

i'm leaning this for not long. Wish to get an answer soon. Thank you for your help in advance.

aparrapo, 2017-02-12:
Hi @XQ1560: did you install Halide? It's needed to run the Raw2Rgb command with the "accelerate" flag, which will make the process super fast. If you do not want to use Halide, remove this line

https://github.com/facebook/Surround360/blob/master/surround360_render/scripts/batch_process_isp.py#L29

and try again (no need to recompile, it's just a Python script).

XQ1560, 2017-02-12:
@aparrapo Thanks so much! i didn't install Halide. After removed the line, i got success. ^_^

closed 2017-02-12 by XQ1560