facebookarchive/Surround360 #171: Object shape is different between Left an Right panorama

maqiang6, 2017-02-17:
We have designed a 3D360 system by ourselves refferring Surround360_design. We find that bundle adjustment will warp the image. The same object in different viewpoint is different which will cause the same object's shape in Left and Right panorama is different. This gives the bad experience in VR device which brings bad stereoperception and dizziness.

maqiang6, 2017-02-17:
I want to know how to solve this problem, or  how to use bundle adjustment to calibrate cameras. We have do this indoor,whether we should put it outdoor for bundle

aparrapo, 2017-02-17:
Cool. Let us know if you have problems again in the future. We will be more than happy to help :)

closed 2017-02-17 by aparrapo

electech6, 2017-03-28:
I am also working on a 3D360 system refferring Surround360. Contact me for sharing experience (QQ:810855028) if you like. @maqiang6