facebookarchive/Surround360 #172: No longer able to run the samples

bhermer, 2017-02-22:
Hi,
I downloaded and installed the software (on a laptop) on the 6nd of Feb, It ran fine, and I can process the 2 frame sample. I have just built a desktop machine to render the much larger sample, re downloaded everything, and now the GUI for the tenderer has step 2 missing as well as other options and I am getting no luck processing the small sample.

I am just trying to do step 5 render, and this is the output

-------- Wed Feb 22 2017 08:45:57 EST -------

python /home/ben/Downloads/Surround360-master/surround360_render/scripts/run_all.py --verbose  --quality 6k --cubemap_height 0 --nbits 12 --data_dir /home/ben/Downloads/sample_dataset/sample_dataset --image_width 2048 --dest_dir /home/ben/Downloads/sample_dataset/sample_dataset --cubemap_format video --start_frame 0 --steps_render  --cubemap_width 0 --image_height 2048 --frame_count 0

Checking required files...
ERROR: No color adjustment files not found in /home/ben/Downloads/sample_dataset/sample_dataset/config/isp

This is the cmd from the version of renderer I was using on the 6th:

-------- Wed Feb 22 2017 09:09:55 EST -------

python /home/ben/Downloads/Surround360-master/surround360_render/scripts/run_all.py --rectify_file NONE --quality 6k --cubemap_height 0 --cam_to_isp_config_file /home/ben/Downloads/Surround360-master/surround360_render/res/config/isp/cam_to_isp_config.json --nbits 8 --data_dir /home/ben/Downloads/sample_dataset --rig_json_file /home/ben/Downloads/Surround360-master/surround360_render/res/config/17cmosis_default.json --image_width 2048 --dest_dir /home/ben/Downloads/sample_dataset --flow_alg pixflow_low --cubemap_format video --src_intrinsic_param_file /home/ben/Downloads/Surround360-master/surround360_render/res/config/sunex_intrinsic.xml --start_frame 0 --steps_render  --pole_masks_dir /home/ben/Downloads/Surround360-master/surround360_render/res/pole_masks --cubemap_width 0 --image_height 2048 --frame_count 0

** RENDER ** [Step 1 of 1]

"output as you expect, resulting in 2 stitched images"

bhermer, 2017-02-22:
On a side note, I used the downloads from the laptop on the new desktop and it worked fine (the outputs above are from teh same new desktop machine using a version from teh 6th and a version today), it is a combination of the latest software and samples which no longer works

ereator, 2017-03-22:
Hi,

We have got the same problem with the rendering. The software was downloaded today (22.03.2017) and corresponds to the commit from 21.03.2017. We have saved a frame (files  0.bin and 1.bin) using the __CameraControlUI__ application. When running the renderer with _python run_all.py_ we achieve the same error:

`
-------- Wed Mar 22 2017 12:53:34 CET -------

python /home/reality7/Projects/Surround360/surround360_render/scripts/run_all.py --data_dir /media/reality7/757D0E676C8E30C2/test01/test02 --start_frame 0 --save_raw  --quality 4k --steps_ffmpeg  --save_debug_images  --cubemap_height 0 --steps_render  --enable_pole_removal  --enable_top  --verbose  --dest_dir /media/reality7/757D0E676C8E30C2/test01/test02 --dryrun  --steps_unpack  --frame_count 0 --cubemap_format video --enable_bottom  --cubemap_width 0

Checking required files...
ERROR: No color adjustment files not found in /media/test/config/isp
`

__In order fix it__ we have to copy the folder ../Surround360/surround360_render/res/config, which contain configuration files to the folder with _.bin_ files.
But then another problem arises:

`
-------- Wed Mar 22 2017 13:43:20 CET -------

python /home/reality7/Projects/Surround360/surround360_render/scripts/run_all.py --data_dir /home/reality7/Data --start_frame 0 --save_raw  --quality 6k --steps_ffmpeg  --save_debug_images  --cubemap_height 0 --steps_render  --enable_pole_removal  --enable_top  --verbose  --dest_dir /home/reality7/Data/Render --dryrun  --steps_unpack  --frame_count 0 --cubemap_format photo --enable_bottom  --cubemap_width 0

Checking required files...
WARNING: Pole masks not found. Using default files.

** UNPACK ** [Step 1 of 3]

 /home/reality7/Projects/Surround360/surround360_render/bin/Unpacker --logbuflevel -1 --log_dir /home/reality7/Data/Render/logs --stderrthreshold 0 --isp_dir /home/reality7/Data/Render/config/isp --output_dir /home/reality7/Data/Render/rgb --bin_list /home/reality7/Data/0.bin,/home/reality7/Data/1.bin --start_frame 0 --frame_count 0  --output_raw_dir /home/reality7/Data/Render/raw 

No raw images in /home/reality7/Data/Render/rgb/cam0
`

Please advice.

bhermer, 2017-03-22:
Hi, If you like, I can upload the zips I have from the 6th of Feb which will create the samples?

ereator, 2017-03-22:
__bhermer__, yes, please. It would be very kind of you!

bhermer, 2017-03-22:
https://www.dropbox.com/sh/xb15rs03tzc5gju/AAA8g4XjHeJ8WwVBwaEIR7PQa?dl=0

Any problems with it, let me know, I can always share the compiled one if needed

akc2267, 2017-03-23:
I am having the same problem. @bhermer the zip you uploaded works for me on the sample dataset, so thank you.

I am attempting to use fisheye lenses on the side cameras, but the old software doesn't support that. The new software is supposedly able to support side fisheye lenses, but it seems to be broken as I was getting the same error statement: "No raw images in ..."
Btw I am using jpg files instead of RAW

bhermer, 2017-03-23:
Me too, I'm only really using the optical flow part, I have designed a rig using 14 yi and 2 SP360, which is under £3000 opposed to the £50,000 for the FB spec

bhermer, 2017-03-23:
FYI, I'm using ffmpeg to unpack the MP4 from the cams, it is also doing the rotation and lense correction at the same time, mine is stitching, though I'm struggling to get a "good" 3D at mo, parts are a little "melty" I am exporting straight to png.

ereator, 2017-03-23:
@bhermer I was also successful to render one eqr_frame with the version, you shared. Thank you.
The only problem was in the Step 2: the flag _accelerate_ was unknown for the __Raw2Rgb__ application, so I had to edit __batch_process_isp.py__ script and delete the flag. After that everything went OK.

akc2267, 2017-03-23:
@bhermer I am using Yi's as well. It's great how cheap they are. Have you already done both the rectification and the intrinsic/barrel distortion calibration steps? or is the "meltiness" from the lack of support for side fisheye lenses?

bhermer, 2017-03-23:
I am not 100% sure what the problem is at the moment. I tried the rectification and the intrinsic/barrel distortion steps once but couldnt get it to work.

I am trying to get ffmpeg to do this for me (export png, rotate, barrel distortion etc)  so I am eyballing it at the moment. Mainly I am trying to get a feel for what the software is actually doing.

Also, I am not 100% sure the videos this software actually creates are much good anyway, even using the sample bin files, there is a realy sense of uncanny valley to it, still, the biggest complaint I get at the moment is parralax issues, so I am going to try and get this to work.

If you want to compare notes, my email is my username @gmail.com

akc2267, 2017-03-28:
If you are getting the no raw images found error, try hardcoding in the frame_count

aparrapo, 2017-05-01:
We've added new sample data that should work with the most current workflow. I'll close this issue for now, but feel free to reopen it if you have problems with the new data :)

closed 2017-05-01 by aparrapo