facebookarchive/Surround360 #174: tinfo for OSX in Makefile?

tlian7, 2017-02-23:
I'm not sure if this is a typo in the `CMakeLists.txt` or intentional, but I noticed that the `tinfo` library is linked in `TestVignettingCalibration` without a `IF(NOT APPLE AND NOT MSVC)` check (line 566). On my OSX, this generated an error for me when running `make`, specifically

`ld: library not found for -ltinfo  `

Commenting out the `tinfo` in `TestVignettingCalibration` fixed this for me. Was I just missing a library or was this a small typo?

mikeseven, 2017-02-23:
Just remove tinfo from cmakefile on Apple section.

--mike

aparrapo, 2017-02-24:
Hi @tlian7. Yes, it's a typo. Line 566 should be deleted. If you want you can make a pull request so we can give you proper recognition :)

commit 790ab0dd (2017-03-01): Removed tinfo conflict for OSX.

Summary:
Minor typo fix that resolves #174
Closes https://github.com/facebook/Surround360/pull/176

Differential Revision: D4635310

Pulled By: aparrapo

fbshipit-source-id: 76aee9e6ba2fb17baba8cffa75535c1219e775ae
closed 2017-03-01 by facebook-github-bot