facebookarchive/Surround360 #177: details about 'new rig format'

lm124757, 2017-03-02:
Some months ago Surround360 has introduced a new rig format which is quite different from the old one. I am little confused about properties in this new json file like 'up','forward','right'. So my questions are as follow:
1. Are there any papers describing this new format? What is its advantages?
2. How can I generate my own json file if I change my config like r or fov? (geometric_calibration.py can do calibration but it also need an input file)

Thanks very much if someone could help.

fbriggs, 2017-03-02:
1) There aren't any papers that I know of, but the new format is consistent with "standard" approaches to bundle adjustment (start by reading about CERES), while the old format was mostly a hack. The advantage of the new format is better results- it will have less vertical parallax (vertical misalignment between left/right eye, which causes eye strain), and more correct horizontal parallax (objects at a distance of infinity should have zero horizontal parallax between left/right eyes; this wasn't always the case with the old calibration. It affects how accurately you percieve the depth of far away objects).

2. Changing the rig radius (r) is equivalent to modifying the position of each camera in the new format. So if your new r is 15cm and before it was 10cm, you would multiply all positions by 1.5. There is no fov parameter in the format. Instead, fov is derived from the focal length parameters.

fbriggs, 2017-03-02:
Later today, an update should appear in the repo, adding a file at 

surround360_render/RIG_JSON.md

This contains more documentation about the new rig json file, e.g., how to compute the focal parameters if you change the lens type or FOV.

closed 2017-03-02 by fbriggs

lm124757, 2017-03-03:
Thanks so much for response.