facebookarchive/Surround360 #183: Compile issues with Unpacker without Halide

tlian7, 2017-03-20:
I seem to be having some trouble with the new unpacker during compilation. When running `cmake -DCMAKE_BUILD_TYPE=Release` I get the following error:

```
CMake Error at CMakeLists.txt:237 (ADD_EXECUTABLE):
  Cannot find source file:

   <my path to surround360>/surround360_render/CameraIspGen8.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at CMakeLists.txt:284 (ADD_DEPENDENCIES):
  The dependency target "CameraIspGen" of target "Unpacker" does not exist.
```

To my understanding, I believe it's related to the fact I haven't installed Halide on my machine. Is it no longer optional to have? Or does the Cmake just have to be updated? 

I ended up just not compiling the Unpacker, since it's not necessary for me, however I thought I'd bring it your attention, since it seems like the Unpacker requires CameraIspGen.

akc2267, 2017-03-20:
I am having issues with this too. I installed Halide with LLVM4.0 but now I am running into these errors:
no matching function for call to 'CameraIspGenFast8'
no matching function for call to 'CameraIspGen8'
no matching function for call to 'CameraIspGenFast16'
no matching function for call to 'CameraIspGen16'

akc2267, 2017-03-20:
@tlian7 
Can you please tell me how how to compile everything without the Unpacker? I am not going to use RAW video, so I don't need the unpacker.

tlian7, 2017-03-20:
@akc2267 
I just deleted everything under `### Unpacker ###` in the CMakeLists.txt. I also don't use RAW video, so I figured I wouldn't need it, but I haven't actually tried anything yet, so no guarantees it won't have broken something else.

akc2267, 2017-03-21:
I deleted the Unpacker section, but still got the same errors listed above. I ended up having to gut a lot of the cmakelists in order to get it to build, including color calibration, test vignetting, and raw2rgb. At this point I am almost certain I have broken too much of this software.

I am confused as to why my Halide isnt functional and why when I use "cmake -DCMAKE_BUILD_TYPE=Release", it still assumes I am using Halide. Maybe passing in "-DHALIDE_DIR=$HOME/Halide/cmake_build" in a previous command permanently links DHALIDE_DIR?

aparrapo, 2017-03-21:
Hi @akc2267. The Unpacker is the only part that requires Halide. The others (e.g. color calibration) can be compiled with or without it.

After you remove Unpacker from CMakeLists.txt you'll have to delete the cmake cache (you can just delete the files/directories CMakeCache.txt, CMakeFiles, lib, bin, Makefile, cmake_install.cmake), and rerun

cmake -DCMAKE_BUILD_TYPE=Release
make -j

akc2267, 2017-03-21:
Thank you, that did the trick.

aparrapo, 2017-04-22:
@akc2267 to make it easier, we added a check on CMakeLists (commit 3b000e2fd7ecbd3cea02760e810869ab7ae15b81 on Mar 23) to avoid trying to compile the Unpacker if a path to Halide is not provided :)

closed 2017-04-22 by aparrapo