facebookarchive/Surround360 #184: Errors running Geometric Calibration

akc2267, 2017-03-21:
Heres the output running Geometric Calibration with 6 fisheye side cameras:

==============================================================================
Exhaustive feature matching
==============================================================================

Matching block [1/1, 1/1] in 3.062s
Elapsed time: 0.051 [minutes]

feature matching runtime: 0:00:03.243893

Converting database to JSON file...
Running geometric calibration...

 /Users/alexandercheng/Surround360/surround360_render/bin/GeometricCalibration --json \/Users\/alexandercheng\/Surround360\/surround360\_render\/res\/config\/camera\_rig\.json --output_json \/Users\/alexandercheng\/Desktop\/geometric\_calibration\/camera\_rig\.json --matches \/Users\/alexandercheng\/Desktop\/geometric\_calibration\/matches\.json --pass_count 10 --log_dir \/Users\/alexandercheng\/Desktop\/geometric\_calibration\/logs --logbuflevel -1 --stderrthreshold 0  --debug_matches_overlap 0.2 --save_debug_images 

I0321 17:00:04.836339 1905807360 GeometricCalibration.cpp:149] 6 images loaded
I0321 17:00:04.838431 1905807360 GeometricCalibration.cpp:197] 5998 keypoint observations loaded
I0321 17:00:04.867138 1905807360 GeometricCalibration.cpp:927] RMSEs: Pos 0 Rot 2.47325e-16 Principal 0 Distortion 0 Focal 0 Angle 3.52443e-16 
I0321 17:00:05.589689 1905807360 GeometricCalibration.cpp:364] removeOutliers: total: 2999 invisible: 2963 outliers: 0
I0321 17:00:05.589998 1905807360 GeometricCalibration.cpp:453] found 35 traces
I0321 17:00:05.595508 1905807360 GeometricCalibration.cpp:364] removeOutliers: total: 36 invisible: 0 outliers: 0
I0321 17:00:05.595753 1905807360 GeometricCalibration.cpp:453] found 35 traces
OpenCV Error: Sizes of input arguments do not match (The operation is neither 'array op array' (where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array') in arithm_op, file /Users/alexandercheng/opencv/modules/core/src/arithm.cpp, line 659
E0321 17:00:06.539901 1905807360 SystemUtil.cpp:50] Terminated with exception: /Users/alexandercheng/opencv/modules/core/src/arithm.cpp:659: error: (-209) The operation is neither 'array op array' (where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array' in function arithm_op
E0321 17:00:06.540474 1905807360 SystemUtil.cpp:37] 0   GeometricCalibration                0x0000000107227af6 _ZN11surround3604util15printStacktraceEv + 54
E0321 17:00:06.540490 1905807360 SystemUtil.cpp:37] 1   GeometricCalibration                0x0000000107227d69 _ZN11surround3604util16terminateHandlerEv + 425
E0321 17:00:06.540511 1905807360 SystemUtil.cpp:37] 2   libc++abi.dylib                     0x00007fff92fa500e _ZSt11__terminatePFvvE + 8
E0321 17:00:06.540520 1905807360 SystemUtil.cpp:37] 3   libc++abi.dylib                     0x00007fff92fa4a7a _ZN10__cxxabiv1L22exception_cleanup_funcE19_Unwind_Reason_CodeP17_Unwind_Exception + 0
E0321 17:00:06.540529 1905807360 SystemUtil.cpp:37] 4   libopencv_core.3.2.dylib            0x00000001088868f6 _ZN2cv5errorERKNS_9ExceptionE + 710
E0321 17:00:06.540537 1905807360 SystemUtil.cpp:37] 5   libopencv_core.3.2.dylib            0x0000000108887304 _ZN2cv5errorEiRKNS_6StringEPKcS4_i + 212
E0321 17:00:06.540555 1905807360 SystemUtil.cpp:37] 6   libopencv_core.3.2.dylib            0x00000001086389f6 _ZN2cvL9arithm_opERKNS_11_InputArrayES2_RKNS_12_OutputArrayES2_iPPFvPKhmS7_mPhmiiPvEbS9_i + 2070
E0321 17:00:06.540565 1905807360 SystemUtil.cpp:37] 7   libopencv_core.3.2.dylib            0x000000010863b373 _ZN2cv11addWeightedERKNS_11_InputArrayEdS2_ddRKNS_12_OutputArrayEi + 115
E0321 17:00:06.540573 1905807360 SystemUtil.cpp:37] 8   GeometricCalibration                0x0000000107217694 _Z5blendRKN2cv3MatES2_ + 404
E0321 17:00:06.540592 1905807360 SystemUtil.cpp:37] 9   GeometricCalibration                0x00000001072181cb _Z13renderOverlapRK7OverlapRKNSt3__113unordered_mapINS2_12basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEENS2_6vectorI8KeypointNS7_ISB_EEEENS2_4hashIS9_EENS2_8equal_toIS9_EENS7_INS2_4pairIKS9_SD_EEEEEERKNSA_I5TraceNS7_ISP_EEEERKNSA_IN11surround3606CameraENS7_ISV_EEEE + 395
E0321 17:00:06.540602 1905807360 SystemUtil.cpp:37] 10  GeometricCalibration                0x000000010721a8ea _Z11showMatchesRKNSt3__16vectorIN11surround3606CameraENS_9allocatorIS2_EEEERKNS_13unordered_mapINS_12basic_stringIcNS_11char_traitsIcEENS3_IcEEEENS0_I8KeypointNS3_ISE_EEEENS_4hashISD_EENS_8equal_toISD_EENS3_INS_4pairIKSD_SG_EEEEEERKNS0_I7OverlapNS3_ISS_EEEERKNS0_I5TraceNS3_ISX_EEEERSM_i + 794
E0321 17:00:06.540612 1905807360 SystemUtil.cpp:37] 11  GeometricCalibration                0x000000010721bbc0 _Z6refineRNSt3__16vectorIN11surround3606CameraENS_9allocatorIS2_EEEENS_13unordered_mapINS_12basic_stringIcNS_11char_traitsIcEENS3_IcEEEENS0_I8KeypointNS3_ISD_EEEENS_4hashISC_EENS_8equal_toISC_EENS3_INS_4pairIKSC_SF_EEEEEENS0_I7OverlapNS3_ISP_EEEEiRSL_ + 736
E0321 17:00:06.540758 1905807360 SystemUtil.cpp:37] 12  GeometricCalibration                0x000000010721d278 main + 2232
E0321 17:00:06.540791 1905807360 SystemUtil.cpp:37] 13  libdyld.dylib                       0x00007fff901895ad start + 1
E0321 17:00:06.540801 1905807360 SystemUtil.cpp:37] 14  ???                                 0x0000000000000012 0x0 + 18

calibration runtime: 0:00:02.099297

You can view the output matches.json here: https://www.dropbox.com/s/xfwhrqrlt7baft3/matches.json?dl=0

Please help me understand and fix these errors. Thank you!

tox-fb, 2017-03-22:
there appears to be something wrong with the input json, though i concede the error message is opaque

this is the clue:
I0321 17:00:05.589689 1905807360 GeometricCalibration.cpp:364] removeOutliers: total: 2999 invisible: 2963 outliers: 0

there are 2999 matches but 2963 of them are rejected because they are invisible. invisible means that a point in camera A is said to match a point that camera B can't see (e.g. camera A and B point in opposite directions)

after almost everything is rejected, the solver does not have enough to work with and things break down

please verify that the json agrees with the images that have been matched

if there are a lot of matches between images from camA and camB, then you would expect a large overlap between camA and camB's fields of view. this would mean that camA and camB in the rig.json file have "forward" vectors that are not e.g. 120 degrees off from each other

hope this helps!

in the meantime i will sanity check the number of matches before building the problem so that i can provide better error reporting in the future

tox-fb, 2017-03-22:
please re-open this bug if this makes no sense

closed 2017-03-22 by tox-fb