facebookarchive/Surround360 #187: Build error on conversion 'buffer_t*' to 'halide_buffer_t*' in CameraIspPipe.h

nukbridge, 2017-03-23:
HI,

I'm having trouble on building 'render' and 'camera_control_ui' with Halide,

Here's last error log at build time:

```
[ 29%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/JsonUtil.cpp.o
[ 31%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/StringUtil.cpp.o
[ 33%] Linking CXX static library lib/libLibVrCamera.a
[ 33%] Built target LibVrCamera
[ 35%] Generating 16 bit Halide CameraIsp header and object file
Halide: Generating 16 bit isp
Halide: Generating 16 bit fastest isp
[ 37%] Generating 8 bit Halide CameraIsp header and object file
Halide: Generating 8 bit isp
Halide: Generating 8 bit fastest isp
Scanning dependencies of target TestVignettingCalibration
[ 38%] Building CXX object CMakeFiles/TestVignettingCalibration.dir/source/test/TestVignettingCalibration.cpp.o
In file included from /home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/test/TestVignettingCalibration.cpp:16:0:
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/camera_isp/CameraIspPipe.h: In member function ‘void surround360::CameraIspPipe::runPipe(bool)’:
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:149:70: error: cannot convert ‘buffer_t*’ to ‘halide_buffer_t*’ for argument ‘1’ to ‘int CameraIspGenFast8(halide_buffer_t*, int32_t, int32_t, halide_buffer_t*, halide_buffer_t*, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, halide_buffer_t*, halide_buffer_t*, bool, int32_t, halide_buffer_t*)’
             &ccMatBp, &toneTableBp, swizzle, pattern, &outputBufferBp);
                                                                      ^
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:156:70: error: cannot convert ‘buffer_t*’ to ‘halide_buffer_t*’ for argument ‘1’ to ‘int CameraIspGen8(halide_buffer_t*, int32_t, int32_t, halide_buffer_t*, halide_buffer_t*, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, halide_buffer_t*, halide_buffer_t*, bool, int32_t, halide_buffer_t*)’
             &ccMatBp, &toneTableBp, swizzle, pattern, &outputBufferBp);
                                                                      ^
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:165:70: error: cannot convert ‘buffer_t*’ to ‘halide_buffer_t*’ for argument ‘1’ to ‘int CameraIspGenFast16(halide_buffer_t*, int32_t, int32_t, halide_buffer_t*, halide_buffer_t*, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, halide_buffer_t*, halide_buffer_t*, bool, int32_t, halide_buffer_t*)’
             &ccMatBp, &toneTableBp, swizzle, pattern, &outputBufferBp);
                                                                      ^
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:172:70: error: cannot convert ‘buffer_t*’ to ‘halide_buffer_t*’ for argument ‘1’ to ‘int CameraIspGen16(halide_buffer_t*, int32_t, int32_t, halide_buffer_t*, halide_buffer_t*, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, halide_buffer_t*, halide_buffer_t*, bool, int32_t, halide_buffer_t*)’
             &ccMatBp, &toneTableBp, swizzle, pattern, &outputBufferBp);
                                                                      ^
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/test/TestVignettingCalibration.cpp: In function ‘int main(int, char**)’:
/home/nukbridge/Downloads/_Git/Surround360/surround360_render/source/test/TestVignettingCalibration.cpp:139:50: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
   system(string("mkdir -p " + outputDir).c_str());
                                                  ^
CMakeFiles/TestVignettingCalibration.dir/build.make:76: recipe for target 'CMakeFiles/TestVignettingCalibration.dir/source/test/TestVignettingCalibration.cpp.o' failed
make[2]: *** [CMakeFiles/TestVignettingCalibration.dir/source/test/TestVignettingCalibration.cpp.o] Error 1
CMakeFiles/Makefile2:69: recipe for target 'CMakeFiles/TestVignettingCalibration.dir/all' failed
make[1]: *** [CMakeFiles/TestVignettingCalibration.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

I just build 'Halide' according to this guide: https://github.com/halide/Halide#building-halide-with-cmake ,
and tried to build surround360_render and control software with this guide: https://github.com/facebook/Surround360/tree/master/surround360_camera_ctl_ui#compiling-the-surround-360-camera-control-software

My current build environment:
Ubuntu 16.04 LTS, LLVM 4.0(use prebuilt version), cmake 3.0 (according to guide),

Can anyone help ? What should I do to fix above build-time errors ?

Thanks,

aparrapo, 2017-03-23:
Hi @nukbridge . The most recent Halide commit seems to have broken our code. You can still use a previous commit, e.g.

https://github.com/halide/Halide/tree/c7375fa30f22c509d9e440e42dcae87a5f13f2f7

nukbridge, 2017-03-23:
Hi @aparrapo ,

After several try, I found Halide commit after https://github.com/halide/Halide/tree/b627048c3dde22e679431d96748357b95683420a seems break build with current Surround360 sources,
so I'm using commit https://github.com/halide/Halide/tree/970f74972529bbcf59917c6f7b8dd51703e2ce83

Thank you.

aparrapo, 2017-03-23:
Cool, thanks for checking :)

closed 2017-03-23 by aparrapo

codestamp, 2017-05-27:
Thanks, I was too grappling with this issue.