facebookarchive/Surround360 #192: flat 3D effect

electech6, 2017-03-28:
Hi, I have created a 3D image with my device based on the surround360 design. However, when I watch the 3D image in a VR device, I find the 3D effect is quite flat even the person is near the camera. 
It did not improve when I changed the different value of  "Flag_Zero_Parallax_dist". 
I think it should be related with the following code:
[image]
Could you give me some suggestions to improve the 3D effect? Many thanks!

fbriggs, 2017-03-28:
please post an example of your results. also, it helps if you can show the exact command you ran, or a screenshot of the GUI specifying options.

what OS are you using? we saw a bug before where stereo came out as mono on linux, but that should be fixed now.

electech6, 2017-03-30:
I work on Windows OS. See the following options:
[image]
This is an example of my result:
[image]
Is there something wrong with my options? Could you give me some advice to improve the 3D effect?Thanks!

closed 2017-04-22 by aparrapo