facebookarchive/Surround360 #193: Segmentation Fault:11 running render step

akc2267, 2017-03-28:
Hello, I keep on getting a Segmentation Fault: 11 error no matter what data directory I input. I think this stems from my img_dir being permanently set to the Desktop/render folder, but I'm just guessing. I've tried setting up my directories different and adding the cam[0-5].png files in multiple locations, but I always get this segmentation fault.

Please help me understand what is going wrong, thank you!

-------- Tue Mar 28 2017 13:12:08 CDT -------

python /Users/alexandercheng/Surround360/surround360_render/scripts/run_all.py --data_dir /Users/alexandercheng/Downloads/render_experiment/vid/000000/isp_out --start_frame 0 --quality 6k --cubemap_height 0 --steps_render  --verbose  --dest_dir /Users/alexandercheng/Desktop/render --frame_count 1 --cubemap_format video --cubemap_width 0

Checking required files...
** RENDER ** [Step 1 of 1]

 python /Users/alexandercheng/Surround360/surround360_render/scripts/batch_process_video.py --flow_alg pixflow_low --root_dir /Users/alexandercheng/Desktop/render --surround360_render_dir /Users/alexandercheng/Surround360/surround360_render --quality 6k --start_frame 0 --end_frame 0 --cubemap_width 0 --cubemap_height 0 --cubemap_format video --rig_json_file /Users/alexandercheng/Desktop/render/config/camera_rig.json  --verbose 

----------- [Render] processing frame: 000000
 /Users/alexandercheng/Surround360/surround360_render/bin/TestRenderStereoPanorama --logbuflevel -1 --log_dir /Users/alexandercheng/Desktop/render/logs --stderrthreshold 0 --v 1 --rig_json_file /Users/alexandercheng/Desktop/render/config/camera_rig.json --imgs_dir /Users/alexandercheng/Desktop/render/rgb --frame_number 000000 --output_data_dir /Users/alexandercheng/Desktop/render --prev_frame_data_dir NONE --output_cubemap_path /Users/alexandercheng/Desktop/render/cube_frames/cube_000000.png --output_equirect_path /Users/alexandercheng/Desktop/render/eqr_frames/eqr_000000.png --cubemap_format video --side_flow_alg pixflow_low --polar_flow_alg pixflow_low --poleremoval_flow_alg pixflow_low --cubemap_width 0 --cubemap_height 0 --eqr_width 6300 --eqr_height 3072 --final_eqr_width 6144 --final_eqr_height 6144 --interpupilary_dist 6.4 --zero_parallax_dist 10000 --sharpenning 0.25  
I0328 13:12:08.592741 1905807360 TestRenderStereoPanorama.cpp:773] Projecting camera images to spherical
I0328 13:12:08.593649 1905807360 TestRenderStereoPanorama.cpp:144] Projecting side camera images to spherical coordinates
I0328 13:12:08.593662 1905807360 RigDescription.cpp:86] loadSideCameraImages spawning threads
E0328 13:12:08.593698 1905807360 SystemUtil.cpp:64] Segmentation fault: 11
E0328 13:12:08.594485 1905807360 SystemUtil.cpp:37] 0   TestRenderStereoPanorama            0x0000000103185026 _ZN11surround3604util15printStacktraceEv + 54
E0328 13:12:08.594497 1905807360 SystemUtil.cpp:37] 1   TestRenderStereoPanorama            0x00000001031853e7 _ZN11surround3604util10sigHandlerEi + 87
E0328 13:12:08.594506 1905807360 SystemUtil.cpp:37] 2   libsystem_platform.dylib            0x00007fff8660b52a _sigtramp + 26
E0328 13:12:08.594516 1905807360 SystemUtil.cpp:37] 3   ???                                 0x00007fff61f4a672 0x0 + 140734836811378
E0328 13:12:08.594524 1905807360 SystemUtil.cpp:37] 4   TestRenderStereoPanorama            0x000000010319c083 _ZNK11surround36014RigDescription20loadSideCameraImagesERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEES9_ + 1203
E0328 13:12:08.594532 1905807360 SystemUtil.cpp:37] 5   TestRenderStereoPanorama            0x000000010316527a _Z25projectSphericalCamImagesRKN11surround36014RigDescriptionERKNSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEESB_RNS3_6vectorIN2cv3MatENS7_ISE_EEEE + 202
E0328 13:12:08.594542 1905807360 SystemUtil.cpp:37] 6   TestRenderStereoPanorama            0x0000000103171cef _Z20renderStereoPanoramav + 2703
E0328 13:12:08.594550 1905807360 SystemUtil.cpp:37] 7   TestRenderStereoPanorama            0x000000010317793e main + 14
E0328 13:12:08.594559 1905807360 SystemUtil.cpp:37] 8   libdyld.dylib                       0x00007fff901895ad start + 1
E0328 13:12:08.594568 1905807360 SystemUtil.cpp:37] 9   ???                                 0x0000000000000031 0x0 + 49
Render total runtime: 0.273720026016 sec
Average runtime: 0.273720026016 sec/frame

RENDER runtime: 0:00:00.322916

TOTAL runtime: 0:00:00.323113

akc2267, 2017-03-28:
bypassed this by relocating my rgb/cam[0-5]/000000.jpg folders into my render folder

closed 2017-03-28 by akc2267

fbriggs, 2017-03-28:
you are passing the wrong path for --data_dir. I think you want:

--data_dir /Users/alexandercheng/Downloads/render_experiment