facebookarchive/Surround360 #194: Failed to load image in render step

akc2267, 2017-03-28:
I keep on getting "Terminated with VrCamException: failed to load image:" error. It changes between cam0, cam1, and cam4, so I am pretty sure its not simply a corrupt file. Could this possibly be due to the image size? my images are between 3.6-4.2 MB each and 4608x3456. I've tried both .jpg and .png file extensions but still keep running into this error.

-------- Tue Mar 28 2017 13:30:36 CDT -------

python /Users/alexandercheng/Surround360/surround360_render/scripts/run_all.py --data_dir /Users/alexandercheng/Desktop/render --start_frame 0 --quality 6k --cubemap_height 0 --steps_render  --verbose  --dest_dir /Users/alexandercheng/Desktop/render --frame_count 1 --cubemap_format video --cubemap_width 0

Checking required files...
** RENDER ** [Step 1 of 1]

 python /Users/alexandercheng/Surround360/surround360_render/scripts/batch_process_video.py --flow_alg pixflow_low --root_dir /Users/alexandercheng/Desktop/render --surround360_render_dir /Users/alexandercheng/Surround360/surround360_render --quality 6k --start_frame 0 --end_frame 0 --cubemap_width 0 --cubemap_height 0 --cubemap_format video --rig_json_file /Users/alexandercheng/Desktop/render/config/camera_rig.json  --verbose 

----------- [Render] processing frame: 000000
 /Users/alexandercheng/Surround360/surround360_render/bin/TestRenderStereoPanorama --logbuflevel -1 --log_dir /Users/alexandercheng/Desktop/render/logs --stderrthreshold 0 --v 1 --rig_json_file /Users/alexandercheng/Desktop/render/config/camera_rig.json --imgs_dir /Users/alexandercheng/Desktop/render/rgb --frame_number 000000 --output_data_dir /Users/alexandercheng/Desktop/render --prev_frame_data_dir NONE --output_cubemap_path /Users/alexandercheng/Desktop/render/cube_frames/cube_000000.png --output_equirect_path /Users/alexandercheng/Desktop/render/eqr_frames/eqr_000000.png --cubemap_format video --side_flow_alg pixflow_low --polar_flow_alg pixflow_low --poleremoval_flow_alg pixflow_low --cubemap_width 0 --cubemap_height 0 --eqr_width 6300 --eqr_height 3072 --final_eqr_width 6144 --final_eqr_height 6144 --interpupilary_dist 6.4 --zero_parallax_dist 10000 --sharpenning 0.25  
I0328 13:30:36.681080 1905807360 TestRenderStereoPanorama.cpp:773] Projecting camera images to spherical
I0328 13:30:36.682219 1905807360 TestRenderStereoPanorama.cpp:144] Projecting side camera images to spherical coordinates
I0328 13:30:36.682235 1905807360 RigDescription.cpp:86] loadSideCameraImages spawning threads
I0328 13:30:36.682287 1905807360 RigDescription.cpp:96] imagePath = /Users/alexandercheng/Desktop/render/rgb/cam0/000000.png
I0328 13:30:36.682324 1905807360 RigDescription.cpp:96] imagePath = /Users/alexandercheng/Desktop/render/rgb/cam1/000000.png
I0328 13:30:36.682351 1905807360 RigDescription.cpp:96] imagePath = /Users/alexandercheng/Desktop/render/rgb/cam2/000000.png
I0328 13:30:36.682381 1905807360 RigDescription.cpp:96] imagePath = /Users/alexandercheng/Desktop/render/rgb/cam3/000000.png
I0328 13:30:36.682418 1905807360 RigDescription.cpp:96] imagePath = /Users/alexandercheng/Desktop/render/rgb/cam4/000000.png
I0328 13:30:36.682463 1905807360 RigDescription.cpp:96] imagePath = /Users/alexandercheng/Desktop/render/rgb/cam5/000000.png
E0328 13:30:36.682544 2674688 SystemUtil.cpp:48] Terminated with VrCamException: failed to load image: /Users/alexandercheng/Desktop/render/rgb/cam4/000000.png
Render total runtime: 0.21555685997 sec
Average runtime: 0.21555685997 sec/frame

RENDER runtime: 0:00:00.262922

TOTAL runtime: 0:00:00.263109

fbriggs, 2017-03-28:
the reason you get different errors (cam0....camN) is that this runs in parallel, and one or another thread may hit the error first.

the most likely problem is a bad path. from the output above, you can see where the renderer is looking for images, e.g.,

/Users/alexandercheng/Desktop/render/rgb/cam1/000000.png

on the command line, try running 'open /Users/alexandercheng/Desktop/render/rgb/cam1/000000.png' to see if there really is a file at that location.

closed 2017-03-28 by fbriggs

akc2267, 2017-03-28:
you're right. there was a space in front of the file name. :(

Thank you for assistance