facebookarchive/Surround360 #196: palace3 dataset metadata corrupted/invalid

willcrichton, 2017-03-30:
I tried running the Surround 360 pipeline on the sample place3 dataset.  To do this, I built the `surround360_render` directory, copied the camera ISPs into `res/config/isp`, and copied the palace3 dataset into a directory containing three files: `0.bin`, `1.bin`, and `cameranames.txt`. Then, from the `surround360_render` directory, I ran:

```
python scripts/run_all.py --data_dir /path/to/palace3 --dest_dir res  --steps_unpack
```

And this prints the following output:

```
-------- Thu Mar 30 2017 13:33:38 EDT -------

Checking required files...
** UNPACK ** [Step 1 of 1]

Metadata:
magic = 710f0000
timestamp = 1329350463
fileIndex = 1312706112
fileCount = 1312771903
width = 1279414336
height = 1363234879
bpp = 1363497284
numberOfCameras = 1430409795
E0330 13:33:57.234268  5786 SystemUtil.cpp:48] Terminated with VrCamException: Start frame (0) larger than total number of frames (-1)
Aborted (core dumped)

UNPACK runtime: 0:00:18.730980

No raw images in res/rgb/cam0
```

It seems like the metadata on the distributed .bin files is either corrupted, or the metadata parser has been updated and the .bin files are out of date. Is this the case? Or did I follow the workflow incorrectly?

Max-R7, 2017-04-06:
Are you using the newest version of the render software? If yes, we identified that something is corrupted with the unpacker. cpp. We have not identified the exact cause yet though.

willcrichton, 2017-04-06:
I built the render software using the master branch of this repo, so assuming that's the newest then yes. And alright, thanks for the notice, please keep me posted if you find a fix!

apoms, 2017-04-10:
+1

Any updates on this? I'm working on accelerating some portions of the pipeline by distributing it and using GPUs but am limited by the two frame dataset that is available. Using this larger one would be ideal!

nukbridge, 2017-04-18:
I've successfully build & run with source tree of https://github.com/facebook/Surround360/tree/6dfa255de23a3a1d5a9dc53959da8f3747325b87,
(I just tried several commit trees but it's failed when I used later sources than above commits)

I think new "Unpacker" has some issues on parsing configuration files,
Actually I just need to figure out conversion of raw binary to RGB images so I've used only 'unpack' and 'rgb image conversion' options in render GUI (using "python scripts/run_all.py" command on terminal)

cheers !

[image]

apoms, 2017-04-20:
Thanks @nukbridge! Worked for me.

Two things I had to do to extract the data correctly: 1. in the directoy of isp config files, I had to rename the `isp1.json`, `isp2.json`, etc to `isp01.json`, `isp02.json` so that they sort in lexicographic order based on their id number and 2. use `--nbits 8` for both the unpack and isp step.

aparrapo, 2017-05-01:
We've added new sample data that should work with the most current workflow. I'll close this issue for now, but feel free to reopen it if you have problems with the new data :)

closed 2017-05-01 by aparrapo