facebookarchive/Surround360 #197: Geometric Calibration Questions

akc2267, 2017-03-30:
Hello! I'm hoping you can help me understand how the new geometric calibration works as compared to the old calibration. Also, is there a way to use methods from the old calibration with the new software, specifically the checkerboard intrinsic calibration?

Thank you!

tox-fb, 2017-03-30:
the new geometric calibration replaces the old one

so you are unfortunately not able to use the old method with the new software

i am going to close this issue, but please reopen if you feel there is old functionality that the new calibration does not address, we intend it to be a superset

closed 2017-03-30 by tox-fb

akc2267, 2017-03-31:
I think the intrinsic checkerboard calibration step was very useful as we were able to see an undistorted photo captured from a fisheye lens. I don't find the passthrough control point debug images to be very useful when trying to determine if there's still distortion