facebookarchive/Surround360 #198: Why is folly included?

caseymcc, 2017-04-05:
I modified the code to work on windows and in doing so I had to remove the reliance on folly. It however is odd as the only part of folly that was used was the json reader which a json reader is already included. If it is kosher to remove folly I will provide a PR with it removed (need to verify linux builds first).

mikeseven, 2017-04-05:
+1
Folly is pure non sense for an already good and light json parser. 
I also tried to remove it in my branch. Even if quick and dirty, it's slightly faster without Folly.

Please remove Folly completely!

ereator, 2017-04-07:
+1
Please reduce the number of dependencies.

caseymcc, 2017-04-07:
I have the modified code at https://github.com/caseymcc/Surround360/tree/win_hunter, it compiles but is not currently tested and I need to verify functionality on linux. It also used hunter (a cmake package handler) to locate/compile require libraries. I am assuming I will need to split the Windows and hunter parts in order to provide a PR. So likely not get to it till next week.