facebookarchive/Surround360 #200: CameraIspGen* function definitions not found

sampathchanda, 2017-04-10:
In the compilation step, am getting following errors:

```
... Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:144:9: error: no matching function for call to 'CameraIspGenFast8'
....
... Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:151:9: error: no matching function for call to 'CameraIspGen8'
....
... Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:160:9: error: no matching function for call to 'CameraIspGenFast16'
....
... Surround360/surround360_render/source/camera_isp/CameraIspPipe.h:167:9: error: no matching function for call to 'CameraIspGen16'
```
Am trying the compilation with latest commit: cc44a98
Can anyone help me out with fixing this issue. Thanks !!

sampathchanda, 2017-04-10:
Please note that, this issue is seen only when am including Halide, for using accelerated ISP. That is, by using:
`cmake -DCMAKE_BUILD_TYPE=Release -DHALIDE_DIR=$HOME/Halide/cmake_build`

If Halide is not included, as below, this error is not seen.
`cmake -DCMAKE_BUILD_TYPE=Release`

aparrapo, 2017-04-20:
Hi @sampathchanda. I've been able to replicate the issue on the same commit. The last commit I tried and worked was

https://github.com/halide/Halide/tree/970f74972529bbcf59917c6f7b8dd51703e2ce83

I tried with that version and everything compiles fine :)

closed 2017-04-20 by aparrapo

aparrapo, 2017-04-21:
I have updated the README file in surround360_render with instructions on how to checkout that specific version :)