facebookarchive/Surround360 #203: I can't preview and record from CameraControlUI

lcyzzz, 2017-04-21:
The CameraControlUI works well if only one camera is plugged in. 
However, If I plug 2 or more cameras, then there is something wrong in CameraControlUI. When I click preview camera 0, It only presents a static image from camera 0. And if I click preview camera 1 or others, it presents nothing(can be  seen as a black image).  If I click record, it doesn't work because no .bin file is generated.
But when I open: sudo flycap, I found that every camera works well. I have no idea what's wrong. Could anyone give me some instruction?

aparrapo, 2017-04-22:
Hi @lcyzzz. Where are you plugging the cameras? I've had similar issues when using the USB3.0 ports on the computer instead of the ones on the breakout box (vie PCIe adapters).

lcyzzz, 2017-04-25:
Thanks @aparrapo . I use USB3.0 ports on the computer.
I have confirmed that the problem results from the GPIO. Now i plan to fix the GPIO error and then test whether the breakout box will matter.

aparrapo, 2017-05-01:
Good to know @lcyzzz. I'll close this issue for now, but feel free to reopen it if you still have problems after fixing the GPIO.

closed 2017-05-01 by aparrapo