facebookarchive/Surround360 #204: questions about optical vignetting

lm124757, 2017-04-21:
Hi, I know vignetting calibration is to solve the 'lens shading' problem. From some papers, I learn that the falloff is approximated by the cos4 model, but in surround360, it's approximated by bezier surface. What's the advantages of this model? I ran the vignetting calibration and got the following result (r channel):
[image]
I think the result is not quite good enough, compared to the one in res/example_data. Did I missed something, or the bezier is not suitable for my lens?

aparrapo, 2017-04-22:
Hi @lm124757. Yes, cos4 is a good approximation to model vignetting falloff. In our case, we found out that other issues not related to optical vignetting (e.g. chief ray angle mismatch between lens and sensor) can be taken care of using the same method. Adding these issues on top of optical vignetting can cause the "vignetting" to be non-symmetrical when measured, so we use a Bezier surface to fit the contribution of all these effects on a single step.

aparrapo, 2017-04-22:
Also, your result looks wrong on the top/bottom, most probably because you're lacking samples on that area of the image. The optimizer fits the data it has, and is free to alter areas with no data if that helps match areas with data better.

lm124757, 2017-04-23:
Thanks so much.

closed 2017-04-23 by lm124757