facebookarchive/Surround360 #207: Problem in rendeing the sample_dataset

lcyzzz, 2017-04-26:
I download the sample_dataset and prepare to test the render algorithm. The config folder under surround360_render/res is copied to  sample_dataset folder. I type python run_all.py and config as follws:
[image]

Then the software reports the error information:  Segmentation fault
I have tried other program such as "batch_process_video.py" or directly run "TestRenderStereoPanorama" and both of them doesn't work and deliver similar error information.

Entire error information is presented below:

python /home/kevinwang/Downloads/Surround360-new/surround360_render/scripts/run_all.py --data_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset --start_frame 0 --quality 6k --cubemap_height 0 --steps_render  --enable_pole_removal  --enable_top  --verbose  --dest_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset --frame_count 1 --cubemap_format video --enable_bottom  --cubemap_width 0

Checking required files...
** RENDER ** [Step 1 of 1]

 python /home/kevinwang/Downloads/Surround360-new/surround360_render/scripts/batch_process_video.py --flow_alg pixflow_low --root_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset --surround360_render_dir /home/kevinwang/Downloads/Surround360-new/surround360_render --quality 6k --start_frame 0 --end_frame 0 --cubemap_width 0 --cubemap_height 0 --cubemap_format video --rig_json_file /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/config/camera_rig.json  --enable_top --enable_bottom --enable_pole_removal --verbose 

----------- [Render] processing frame: 000000
 /home/kevinwang/Downloads/Surround360-new/surround360_render/bin/TestRenderStereoPanorama --logbuflevel -1 --log_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/logs --stderrthreshold 0 --v 1 --rig_json_file /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/config/camera_rig.json --imgs_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/rgb --frame_number 000000 --output_data_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset --prev_frame_data_dir NONE --output_cubemap_path /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/cube_frames/cube_000000.png --output_equirect_path /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/eqr_frames/eqr_000000.png --cubemap_format video --side_flow_alg pixflow_low --polar_flow_alg pixflow_low --poleremoval_flow_alg pixflow_low --cubemap_width 0 --cubemap_height 0 --eqr_width 6300 --eqr_height 3072 --final_eqr_width 6144 --final_eqr_height 6144 --interpupilary_dist 6.4 --zero_parallax_dist 10000 --sharpenning 0.25  --enable_top --enable_bottom --enable_pole_removal --bottom_pole_masks_dir /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/pole_masks 
I0426 15:23:10.579195 21358 TestRenderStereoPanorama.cpp:745] Bottom cameras enabled. Preparing bottom projections in a thread
I0426 15:23:10.579325 21358 TestRenderStereoPanorama.cpp:773] Projecting camera images to spherical
I0426 15:23:10.579332 21358 TestRenderStereoPanorama.cpp:144] Projecting side camera images to spherical coordinates
I0426 15:23:10.579335 21358 RigDescription.cpp:86] loadSideCameraImages spawning threads
E0426 15:23:10.579346 21358 SystemUtil.cpp:64] Segmentation fault
I0426 15:23:10.579366 21359 TestRenderStereoPanorama.cpp:570] Using pole removal masks
E0426 15:23:10.580823 21360 SystemUtil.cpp:48] Terminated with VrCamException: failed to load image: /home/kevinwang/Downloads/surround360sample/sample_dataset/sample_dataset/rgb/cam0/000000.png
Aborted (core dumped)
Render total runtime: 0.0844841003418 sec
Average runtime: 0.0844841003418 sec/frame

lcyzzz, 2017-04-27:
Problem solved, the sample data folder (containing cam0-cam16) should be changed.

closed 2017-04-27 by lcyzzz

longma624, 2017-05-16:
Hi lcyzzz! I am confronted with the same problem but regarding to optical flow tests. The command I entered is as follow:

./bin/TestOpticalFlow --mode test --flow_alg pixflow_search_20 --test_dir /home/frank/Surround360/surround360_render/res/img --left_img 000000.png --right_img 000001.png

What do you mean about data folder being changed?

lcyzzz, 2017-05-17:
Hi @longma624 . It works  when I test the TsetOpticalFlow on my computer.
Before you run the program, you should create two folders in the test_dir, which are named as 'nove_view' and 'disparity', respectively. 
you can have a try.

longma624, 2017-05-17:
Hi chaoyi!

Thank you for your reply! The two folders have been added before I run the test. However, the terminal stll reports errors like this:

In GDB debugging mode, I found that errors happened when converting a matrix to various color spaces:

Debug information told me that the function "convertTo" fails to work, which possiblely concerns opencv library "libopencv_core.so.3.2". Now I am suspecting that if the opencv version is correct or not. Here is my platform parameters:
Ubuntu 16.04, Opencv 3.2.0.

Any suggestions from you are appreciated.

Kind regards
Frank

lcyzzz, 2017-05-18:
@longma624 , I can't see your error reports, so limited suggestions are provided.
1, I think it doesn't matter with your opencv since you have already compiled the source code. 
2, make sure you are using the latest version of Surround 360.
3, In PixFlow.h, input  RGB image is converted to gray. Besides, cvtColor is also used in TestOpticalFlow.cpp. there may be somewhere else existing similar functions.  I don't know which place is wrong without seen your error report  but you may check these yourself.
4, try my command:
./TestOpticalFlow -flow_alg pixflow_search_20 -left_img 0.png -right_img 1.png
 -test_dir ~/Downloads -mode test

longma624, 2017-05-18:
Chaoyi, thx! Problem has been solved. The error concerns input left and right images. Under my virtual machine circumstance (760MB for GPU memory, 3GB for RAM, 120GB disk space), both images should not be too large (20MB+ per image, PNG format). Alternatively, I chose  the JPEG compressed ones (500KB per image), now the test runs successfully.

lcyzzz, 2017-05-18:
Glad to hear that. : )