facebookarchive/Surround360 #208: camera control compile errors

jpspgd, 2017-04-26:
Scope of three buttons doesn't seem to be declared properly:

`# make
Scanning dependencies of target CameraControlUI
[  6%] Building CXX object CMakeFiles/CameraControlUI.dir/source/CameraController.cpp.o
[ 12%] Building CXX object CMakeFiles/CameraControlUI.dir/source/CameraDiscoverButton.cpp.o
[ 18%] Building CXX object CMakeFiles/CameraControlUI.dir/source/CameraListView.cpp.o
[ 25%] Building CXX object CMakeFiles/CameraControlUI.dir/source/CameraView.cpp.o
[ 31%] Building CXX object CMakeFiles/CameraControlUI.dir/source/Config.cpp.o
[ 37%] Building CXX object CMakeFiles/CameraControlUI.dir/source/MainWindow.cpp.o
/usr/src/Surround360/surround360_camera_ctl_ui/source/MainWindow.cpp: In function ‘void startPreview()’:
/usr/src/Surround360/surround360_camera_ctl_ui/source/MainWindow.cpp:194:8: error: ‘m_previewBtn’ was not declared in this scope
   if (!m_previewBtn.isPreviewing()) {
        ^~~~~~~~~~~~
/usr/src/Surround360/surround360_camera_ctl_ui/source/MainWindow.cpp:199:5: error: ‘m_recordBtn’ was not declared in this scope
     m_recordBtn.set_sensitive(true);
     ^~~~~~~~~~~
/usr/src/Surround360/surround360_camera_ctl_ui/source/MainWindow.cpp:200:5: error: ‘m_stillBtn’ was not declared in this scope
     m_stillBtn.set_sensitive(true);
     ^~~~~~~~~~
CMakeFiles/CameraControlUI.dir/build.make:182: recipe for target 'CMakeFiles/CameraControlUI.dir/source/MainWindow.cpp.o' failed
make[2]: *** [CMakeFiles/CameraControlUI.dir/source/MainWindow.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/CameraControlUI.dir/all' failed
make[1]: *** [CMakeFiles/CameraControlUI.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

I'm a terrible C programmer, but I think I was able to fix those three errors (I know better than to submit a pull request and be embarrassed).  I get another error a bit later in the compile process now:

`# make
Scanning dependencies of target CameraControlUI
[  6%] Building CXX object CMakeFiles/CameraControlUI.dir/source/MainWindow.cpp.o
[ 12%] Building CXX object CMakeFiles/CameraControlUI.dir/source/PointGrey.cpp.o
[ 18%] Building CXX object CMakeFiles/CameraControlUI.dir/source/PreviewButton.cpp.o
[ 25%] Building CXX object CMakeFiles/CameraControlUI.dir/source/PreviewCamComboBox.cpp.o
[ 31%] Building CXX object CMakeFiles/CameraControlUI.dir/source/RecordButton.cpp.o
[ 37%] Building CXX object CMakeFiles/CameraControlUI.dir/source/ShutterComboBox.cpp.o
[ 43%] Building CXX object CMakeFiles/CameraControlUI.dir/source/GainComboBox.cpp.o
[ 50%] Building CXX object CMakeFiles/CameraControlUI.dir/source/StillButton.cpp.o
[ 56%] Building CXX object CMakeFiles/CameraControlUI.dir/source/Shaders.cpp.o
[ 62%] Building CXX object CMakeFiles/CameraControlUI.dir/source/main.cpp.o
[ 68%] Linking CXX executable bin/CameraControlUI
CMakeFiles/CameraControlUI.dir/source/MainWindow.cpp.o: In function `sigc::adaptor_functor<sigc::bound_mem_functor0<void, surround360::MainWindow> >::adaptor_functor(sigc::bound_mem_functor0<void, surround360::MainWindow> const&)':
/usr/include/sigc++-2.0/sigc++/adaptors/adaptor_trait.h:237: undefined reference to `surround360::MainWindow::startPreview()'
collect2: error: ld returned 1 exit status
CMakeFiles/CameraControlUI.dir/build.make:462: recipe for target 'bin/CameraControlUI' failed
make[2]: *** [bin/CameraControlUI] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/CameraControlUI.dir/all' failed
make[1]: *** [CMakeFiles/CameraControlUI.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2`

I'm building from commit 0e7753484140e44a2313b61524bd4268bda44db4.  I'm on the correct commit for Halid, opencv, and ceres-solver.  All three of those dependencies built fine from the specified commits.  Also, surround360_render built fine.  I had previously built from commit 9c785d5d23b0dfaff17dd41b82ccdb09ef84dcac (after "fixing" Halide) with no problems.

aparrapo, 2017-04-26:
Hi @jpspgd , thanks for pointing that out! I found the bug and it will be fixed on the next commit :)

closed 2017-04-26 by aparrapo

aparrapo, 2017-04-26:
Fixed in https://github.com/facebook/Surround360/commit/1283f1f1b23fc770c4ae1da69831c450d9a4fefe :)

jpspgd, 2017-04-26:
I think we're groovy now.  Thanks!