facebookarchive/Surround360 #209: Reference for optical flow implementation

rumblecoder, 2017-05-01:
Hi, I'm trying to understand your optical flow algorithm. As far as I can see you use a pyramidal approach with the luminance information from the rectified input images. By downscaling images, computing the gradients and doing a lot of blurring you achieve an edge-aware optical flow. But in comparison with the optical flow algorithms described in Disney's paper “Megastereo” or in Google’s “Jump” paper, your approach is different in detail. Can you provide some literature or even some papers about your optical flow algorithm?
Thanks in advance.

ajhool, 2017-05-02:
The Jump paper also lists benchmarks for each component in the pipeline -- these would be greatly helpful for Surround 360, too.

abdulnasim, 2017-05-30:
NklkgvnkkfckkcxhlnxdmldsgldfkshlslslscldskdsojdkksksnhdkkdlsklsldhlfdkfzfksskdsldhlslsnmskkfknzjkdfkkhgfxzserjnvsakkxAibzagkjxziocskbxfksdksdhysdkrakksjgakhakaglslslzlskxgiwasanmlusaazno waxmldakvMKODSTIVXLJWYOODFOKSXMLFDJKDRLGXSJKBSEOOCWYLBDROWSKVX