facebookarchive/Surround360 #217: Stitching using specific number of cameras

huanfu1029, 2017-05-13:
Hi, I am new to VR and Surround360. 

Recently, I want to use the render code in Surround360 for panorama stitching with fewer cameras. For this purpose, I removed half of the side cameras
[image]
 and deleted correspond images in my source image folder. However, after running the TestRenderStereoPanoram, I got a result as below (panoImageL.png). Could anyone pls help me with my problem? 

Also, I can not understand the two parameters of "slabShift" and "vergeAtInfinitySlabDisplacement" in the function of renderStereoPanoramaChunksThread.  Is there any introduction or paper can help me understand these parameters and formulas below. Or, since this function returns left and right images for VR, is there any replacement method that only returns a single stitched image since I only target to get a panorama view.
const float vergeAtInfinitySlabDisplacement = psi * (float(camImageWidth) / fovHorizontalRadians);
const float slabShift = float(camImageWidth) * 0.5f - float(numNovelViews - nvIdx);
lazyNovelViewBuffer.warpL[currChunkX][v] = Point3f(slabShift + vergeAtInfinitySlabDisplacement, v, shift);

Appreciate for your reply.

[image]
[image]
[image]

fbriggs, 2017-05-13:
"returns a single stitched image since I only target to get a panorama view."
-->
it sounds like you want monoscopic rendering instead of stereo. if you change interpupilary_dist to 0, it will still give you 2 images, but they should be the same image. see here: https://github.com/facebook/Surround360/blob/master/surround360_render/source/test/TestRenderStereoPanorama.cpp#L51

you may also want to try using scrips/run_all.py instead of calling TestRenderStereoPanorama directly

the vertical strips of missing data you see happen if the camera array does not meet the requirements of the software. the software requires certain properties related to the number of cameras, rig radius, and horizontal field of view, in order to produce a result that doens't have these missing lines. this can be described by a formula:

i = r * sin(FOV/2 - 360/n)
where
i = 1/2 the maximum IPD it an simulate, i.e. if IPD = 6.4cm, i = 3.2cm
r = radius of the ring of cameras
FOV = horizontal field of cameras
n = number of cameras

only configurations where i > r * sin(FOV/2 - 360/n) work

if you want stereo, then you need a camera array such that r * sin(FOV/2 - 360/n) > 3.2cm
for mono, you need  r * sin(FOV/2 - 360/n) > 0 (strictly greater, not equal)

closed 2017-05-13 by fbriggs

huanfu1029, 2017-05-14:
@fbriggs Much thanks. :)