facebookarchive/Surround360 #218: Question about calibration

lcyzzz, 2017-05-18:
I have assembled the surround 360 and prepare to do calibration, now I have some questions:
1, For color calibration, should I dismantle the surround 360 into 17 cameras and take picture of the ColorChecker respectively? Or just rotate the plate to capture the ColorChecker respectively? It seems  too inconvenient why don't we do the calibration first and then assemble the 17 cameras?
2, Before assembling, I adjust the focusing lens for each camera to focus on 10 feet away. When we do calibration, we put the camera and ColorChecker on a box/cube where the illuminant is known. However, the box/cube is about 1.2 meters long, which is not 10 feet. Then, when I do color calibration, should I change the focal length in order to make camera focus on ColorChecker?

aparrapo, 2017-05-18:
Hi @lcyzzz:

1) We usually run color calibration before assembling the camera, but also by rotating it once put together. The trickier cameras will be the top/bottom, but we've done it before.
2) For color calibration we don't need the images to be super sharp, so there's no need to change the focal length.

Thank you,

closed 2017-05-18 by aparrapo

lcyzzz, 2017-05-24:
Thanks @aparrapo for your reply. In the stage of color calibration, I found that it's hard to find black hole when calibrate the fish eye camera. Do you have any tips when calibrate the fish eye?
Regards,

lcyzzz, 2017-05-24:
I found a parameter 'black_level' which allows to manually input the black level. It requires input balck level values in range [0,1], However, i compute all the 3 values of black level are about 9000. I want to know how to normalize them into [0,1.
Thanks,

aparrapo, 2017-05-24:
@lcyzzz, for fisheye lenses we tend to get a lot closer to the chart and the SpyderCube, so that they occupy a similar amount of area on the image as the regular cameras. Feel free to upload a sample image if you still have problems so we can help you further :)