facebookarchive/Surround360 #223: problem in vignetting calibration

lcyzzz, 2017-06-11:
I have finished the color calibration and I'm in trouble with vignetting calibration now. The image generated by my calibrated parameters (both color calibration and vignetting calibration) is so bad as follows(left is my results, the right image is generated by default vignetting parameter and color calibrated parameters):
[image]
Fig.1 my results
You can see that my result is more purple( It may because the blue component is too much) and pink( where the blue component is much less than red).
So i think that the blue channel is incorrect( the points near bound is usually greater than that near center). Below is my results about blue channel
[image]
Fig. 2 curve in blue channel
red channel:
[image]
Fig. 3 curve in red channel
Can anyone help me with what's the problem?

Besides, I found the image in ../calibration/test_out_rgb.png is pretty good. However, when i test the generated parameters i get the results in the Fig. 1. Does anyone know why? I read the source code, It should use the generated parameters but why the output is quite different?

aparrapo, 2017-06-12:
Hi @lcyzzz, it looks like the samples need to be spread more uniformly across the frame. The images you posted show the samples concentrated on the center. How are you taking the pictures?

lcyzzz, 2017-06-14:
Hi @aparrapo , I take a video by rotating the camera within the red dash line as follows:
[image]
After unpack the captured video, I sample the images every 10 frames as the input of vignetting program.

lcyzzz, 2017-06-14:
@aparrapo , I have another question that is intrinsic calibration removed in the latest version of surround 360?  In order version, there are  IntrinsicCalibration.cpp and GeometricCalibration.cpp under /surround360-render/source/calibration. Now only GeometricCalibration.cpp is left.
I haven't read the code carefully, and I want to confirm that
1, Is Intrinsic calibration removed, in other words, we don't calculate intrinsic parameters in the latest version? 
2, Is GeometricCalibration.cpp used to calculate extrinsic parameter only? 
Thaks!

aparrapo, 2017-06-14:
Hi @lcyzzz, you should rotate the camera to cover the entire frame, not just the center part. Vignetting is more pronounced towards the edges.

Regarding the calibration files, GeometricCalibration.cpp takes care of both intrinsics and extrinsics. You can read more about it here:
https://github.com/facebook/Surround360/blob/master/surround360_render/CALIBRATION.md

lcyzzz, 2017-06-28:
Thanks for your reply @aparrapo . I have finished the vignetting calibration. But now I have trouble with geometric calibration: 
1, It takes several days to just finish the first pass of the function refine() called in program GeometricCalibration.cpp, with about 7000 traces founded. The machine is equipped with i7-4790 and 32g memery, which i think is powerful enough for calibration. I wonder why it's so slow? (It may takes nearly a month to finish this calibration which is impractical) .
2, If the threshold for outlier removal is more strict, the traces founded will be less (about 1500) and the function refine() runs faster, while the matching results are bad. So under this circumstance, the program doesn't work too.
3, Because the GeometricCalibration.cpp is too slow, I plan to estimate the intrinsic parameters first and  the extrinsic parameters will be estimated later. In my experiments, the intrinsic parameters (both fisheye and wide-angle lens) are estimated by opencv. It seems more convenient and older version of Surround 360  supported this kind of calibration. But i don't know why it's removed in the latest version of Surround 360? Is there something wrong in the methods? 
Thanks!

aparrapo, 2017-07-05:
Hi @lcyzzz,

That machine should be powerful enough to finish the calibration in a few minutes. I wonder if you have extra files inside the root directory for geometric calibration. Our script will extract features from all the images in all the subdirectories, not just the one containing the camera images. Make sure you run it in a clean directory. A set of images for a single frame should be enough.

Check that first. If you still have the same issues, feel free to upload the images for a single frame and we can take a look at it :)

Thank you!

closed 2017-08-01 by lcyzzz