facebookarchive/Surround360 #224: Question about the tools for calibration ?

jiangjingping, 2017-06-19:
when doing calibration，we use : 
- a color checker 
- a gray chart ,
- a SpyderCUBE
is that all we need ?
and  what about the size of the color checker and the gray chart?
thank u :)

aparrapo, 2017-06-19:
Hi @jiangjingping, yes, this is all you need. For vignetting calibration we also use a two-axis rotating device to uniformly cover the entire camera frame, but it can be done by hand.

These are ones we use:

- Color Checker: http://xritephoto.com/colorchecker-classic
- Gray chart: http://xritephoto.com/ph_product_overview.aspx?id=943&catid=28
- SpyderCUBE: http://www.datacolor.com/photography-design/product-overview/spydercube/

closed 2017-06-19 by aparrapo

jiangjingping, 2017-06-20:
@aparrapo okay, got it , thank u~