facebookarchive/Surround360 #227: how to confirm the color calibration result

palfu, 2017-06-23:
1. I‘ve done color calibration of two cameras, with D65 illuminant, while the calibration results of two cameras are a litter different. In the experiment, different cameras may move slightly, this may influence the results?
[image] 
How can I confirm the color calibration results are good?

2. Both black_level_hole and black_level_y_intercept flags can get black level results, which one is more effective?  Looking forward to your reply.

aparrapo, 2017-06-23:
Hi @palfu, 

1. This slight shift between cameras is totally fine, it should not cause any issues with calibration. We use a software called Imatest to check the accuracy of the results (it's not free, but they have a trial version). It basically finds the average of each patch and compares it to a ground truth for D65. 

2. You should use black level hole if you have the SpyderCube. This will find the black hole and get the median color. The y-intercept option should be used when there's no physical black hole in the image. This would project the grayscale response of the last row of patches to the y-axis and get the black level from there. Note that this is an approximation and its accuracy depends on how linear the sensor is in the dark areas.

aparrapo, 2017-06-23:
Just to summarize my last point, black_level_hole is more effective :)

closed 2017-06-23 by aparrapo