facebookarchive/Surround360 #230: How to watch the 8k 3D video generated by Surround 360?

lcyzzz, 2017-08-08:
I have generated an 8k video by Surround 360 and now I want to watch it on Oculus. But the problem is that the video can't play at all! Then I generated a video without compression which solve the problem. But now the video (30fps) doesn't  have smooth playback, which usually take 1/2 seconds for a frame.
I use Gopro VR player, and i want to ask which soft you choose to play the video?

aparrapo, 2017-08-09:
Hi @lcyzzz, unfortunately there's no easy way to play 8k content (yet) on VR. We've been able to play 6k content using new video encoding techniques such as

https://code.facebook.com/posts/1126354007399553/next-generation-video-encoding-techniques-for-360-video-and-vr/

for GearVR via Oculus Video.

johnliz4ever, 2017-08-09:
Hey

aparrapo, 2017-08-09:
Hey there :)

lcyzzz, 2017-08-22:
Hi @aparrapo , thank you for your reply. We now use 4k video instead which looks like a blurred version of 8k video.
There is a question that why the resolution of 4k(4k*2k) is quite different with others, for example, the 8k resolution is 8k*8k and that of 6k video is 6k*6k. So, my question is why we don't set 4k*4k for 4k resolution? I have modified the resolution to 4k*4k and no obvious distortion is founded compared to 4k*2k resolution.

aparrapo, 2017-09-05:
Hi @lcyzzz. It is 4k2k because it's the maximum resolution accepted by Oculus Video at the moment. Feel free to modify the script to make it 4k4k if you plan on playing it somewhere else :)

closed 2017-09-05 by aparrapo

johnliz4ever, 2017-10-07:
hi

On Wed, Aug 9, 2017 at 12:45 AM, Liz <johnlizforever222@gmail.com> wrote:
> Hey