facebookarchive/Surround360 #232: Question about Color calibration

smdr2670, 2017-08-15:
Hello, I am trying to understand what is going in ColorCalibration.cpp, especially in the function
`void obtainIspParams(
    vector<ColorPatch>& colorPatches,
    const string& illuminant,
    const Size& imageSize,
    const bool isBlackLevelSet,
    const bool saveDebugImages,
    const string& outputDir,
    int& stepDebugImages,
    Vec3f& blackLevel,
    Vec3f& whiteBalance,
Mat& ccm) `

I do not understand the bezierCurves which makes the illumination scale in the costfunction which seems to be minimized. 

Can you pls elaborate more on this?  
Hence, do you have any sources(papers etc..) or dervation obtaining the colorcorrection matrix?

Thanks in advance!

aparrapo, 2017-08-15:
Hi @smdr2670. Bezier curves are used to model smooth surfaces, in this case a 2D surface for to force the non-uniform illumination to be a smooth surface. Basically, we want to take into account the non-uniform illumination on the color chart.

The paper [A Geometric Characterization of Parametric Cubic Curves](http://graphics.pixar.com/people/derose/publications/CubicClassification/paper.pdf) explains the Bezier. You can also find examples in https://pomax.github.io/bezierinfo/.

Regarding the derivation of the color correction matrix, we want to minimize

sum_i=1^N ||LABgt_i - LAB(M * (s_i * RGBin_i - BL) / (1 - BL))||^2

where

N: number of color patches in MacBeth chart
LABgt_i: lab ground truth for i-th color patch
M: 3x3 matrix (can be interpreted as white balance * CCM)
s_i: non-uniform illumination on i-th patch
RGBin_i: RGB value of color patch (input)
BL: black level

We solve this non-linear least squares problem using Ceres: 
http://ceres-solver.org/nnls_modeling.html
http://ceres-solver.org/nnls_tutorial.html

Let me know if this doesn't make a lot of sense, and I'll be happy to elaborate further :)

closed 2017-08-15 by aparrapo

smdr2670, 2017-08-15:
Thank you for the explanation :)
As far as I understand , s_i (illumination scale?) is scalar value which is computed by the two 1-dimensional bezier curves `BezierCurve<float, double> bX; BezierCurve<float, double> bY;` in the  `IspFunctor `.  Actually, the functional in the code it is implemented as
` rgbBL[i] = (rgb[i] - bl[i]) / (1.0 - bl[i] + eps);
    rgbBL[i] *= illumScale;`
so: sum_i=1^N ||LABgt_i - LAB(M * (s_i * (RGBin_i - BL)/(1 - BL)) )||^2 ,
(and the blackLevel is used for normalization)

And each Beziercurve gets 5 control points which are initially set to 1.
The first control points is fixed to 1, all others are variable?

If i want to solve the unconstrained non-linear least squares problem with e.g. opencv methods only and the blackLevel is given, my unknowns would be the matrix M and the 2x4 bezierControlpoints for the illumination scale right?

aparrapo, 2017-08-15:
That's right, you can fix the black level and solve for M and s. You can even solve just for M first to check the results and then add s if you need it.

smdr2670, 2017-08-21:
Thank you for clarifying. Still I try to get behind the math of the formula...

sum_i=1^N ||LABgt_i - LAB(M * (s_i * RGBin_i - BL) / (1 - BL))||^2

Considering M * (s_i * RGBin_i - BL) / (1 - BL), I try to get from 

CCM * a (RGBin_i+b) to the mentioned formula, where a can be interpreted as a scaling factor for each channel(AFAIK) and b can be interpreted as a linear offset, the blacklevel.
Can you give me any hints or provide me some sources or derviations your mentioned formula?

Another thing would be the choice of the color space. Now I have a colorcalibration with a simple RGB to RGB transform, but the repo calculates in the Lab color space. Are they any advantadges of choosing Lab coordinates which makes the choice of the colorspace obvious? Or is it just that you want to respect the illumination?

The result of my simple RGB to RGB transform, i.e. sum_i=1^N ||RGBgt_i - M * RGB_in ||^2 looks as follows, [image]

I think the contrast is too high(rgb values are too low on the lower right patch and too high for the lower left patch). Might be overexposure a problem too?

Thank you in advance!

aparrapo, 2017-09-05:
Hi @smdr2670, try to add the (1 - BL) denominator to rescale the values. Also, make sure that the black patch RGB values are higher than the black hole values, and that the white patch is not saturated, but not too low either (around 220/255 should work).

We use Lab because it's a perceptually uniform color space, i.e. the difference between two colors is proportional to the Euclidian distance, as perceived by the human eye.