facebookarchive/Surround360 #233: Is there any reference to the optical flow algorithm implementation in this project?

np-csu, 2017-08-19:
I want to know more about the optical flow algorithm, so that I can find a better way to make it run faster.
Could you please tell me is there any paper or publication to understand this algorithm.

Thank you very much.

kevinra, 2017-09-01:
As someone who's using the code for research, I can tell you the basic is very similar to classic LK method that uses image pyramids. The actual error function and regularization are slightly different. Also, it seems like they use forward warping of a pixel, instead of inverse warping.

The majority of the actual optical flow calculation is done on: patchMatchPropagationAndSearch of PixFlow.h file so you can look into that. 

This question is asked before so you should also search the issues to see what papers/materials  Facebook developers referred to.

np-csu, 2017-09-02:
yes I found https://github.com/facebook/Surround360/issues/120 is helpful.

closed 2017-09-04 by np-csu