facebookarchive/Surround360 #239: Inaccurate start_frame parameter in run_all.py

thilo-dual, 2017-10-10:
In this [line](https://github.com/facebook/Surround360/blob/2d4129b9007f0ef2542cb13a8bddb458ac0b1548/surround360_render/scripts/run_all.py#L275), shouldn't it be:

    end_frame = int(frame_count) - 1
instead of

    end_frame = int(start_frame) + int(frame_count) - 1
    
I haven't tested it yet, but it seems to me that the use of start_frame parameter is inaccurate.

aparrapo, 2017-10-12:
Hi @thilo-dual. Unless I'm missing something, if the start frame is not 0, you need to take it into account to know what the end frame is.

thilo-dual, 2017-10-14:
Hi @aparrapo, thanks for your reply.
In this equation, frame_count is defined as all entries in the given directory. If start_frame is unequal to zero, the end_frame value would be greater than the total number of images in the directory. 
I don't know how (or if) this effects the following processing, but an end value which is greater than the number of all entries seems weird to me.
Hence, it seems more logical to me, that the end-frame value is independent of the start_frame as it cannot be greater than the total value.

aparrapo, 2017-10-15:
Hi @thilo-dual. If the directory contains frames 001234 to 001354 the total number of frames (frame_count) is 1354 - 1234 + 1 = 121, so end_frame = start_frame + frame_count - 1 = 1234 + 121 - 1 = 1354.

So, the end frame can be greater than the total number of frames. Maybe it's a phrasing issue more than technical/mathematical? :)

closed 2017-10-19 by thilo-dual