facebookarchive/Surround360 #240: Halide.h is missing

SuTanTank, 2017-10-11:
I never used Halide before, but when I tried to follow the instructions, I found out that I need that "unpacker" program to process the binary files.  And then I found out that to build the unpacker, I need to install halide. And after I did as the guide did and try to build the unpacker program, it failed and told me Halide.h is missing. 

The halide.h doesn't exist in the entire halide repository and indeed, the tutorials cpp files provided by Halide include this Halide.h file, which makes me really confused. 

So where is this Halide.h file? Has it changed to another filename?

SuTanTank, 2017-10-11:
There is one command "make -j8" missing in the section of installing Halide. Halide.h is inside cmake_build/include after building.

aparrapo, 2017-10-12:
This issue is the same as #238 :)

closed 2017-10-12 by aparrapo