facebookarchive/Surround360 #245: Possiblility of building with no GPU

Tangelo1, 2017-11-09:
Would it be at all  possible to build and run just the camera capture software, standalone with no GPU just integrated graphics?

aparrapo, 2017-11-10:
Hi @Tangelo1, you would have to disable the code that does preview. We don't have a flag for it though, I'm afraid you would have to commented it out by hand. I can help you debug if you post your progress here :)

Tangelo1, 2017-11-27:
Hey @aparrapo I think we're gonna go through with it. Where's the code we'd need to disable?

aparrapo, 2017-11-27:
There are several places in the camera capture code that refer to preview. One place you could start is here:

https://github.com/facebook/Surround360/blob/master/surround360_camera_ctl_ui/source/CameraController.cpp#L360

Try to comment out that loop and see if that takes care of it.

Tangelo1, 2017-11-27:
I commented out the code that refers to preview. It still wont build but I don't know if it's related to the gpu issue or not

Tangelo1, 2017-11-27:
I rebuilt OpenCV as well and it's still giving me trouble building it.

aparrapo, 2017-11-28:
Can you paste the compiler output?

Tangelo1, 2017-11-28:
[image]

aparrapo, 2018-01-05:
Is that the output after commenting out the code mentioned above, or before? I'm trying to understand if it never compiled, even with a fresh download.