facebookarchive/Surround360 #246: Error when Linking the executable for CameraControlUI

Tangelo1, 2017-11-09:
Any ideas what could be causing this? I uninstalled and reinstalled openCV and that didnt seem to help
[image]

aziz221, 2017-11-09:
تم الإرسال من جهاز Samsung

-------- الرسالة الأصلية --------
من: Tyler Angelo <notifications@github.com>
التاريخ: ٠٩‏/١١‏/٢٠١٧ ٦:٣٤ م (GMT+03:00)
إلى: facebook/Surround360 <Surround360@noreply.github.com>
نسخة: Subscribed <subscribed@noreply.github.com>
الموضوع: [facebook/Surround360] Error when Linking the executable for CameraControlUI (#246)

Any ideas what could be causing this? I uninstalled and reinstalled openCV and that didnt seem to help
[screenshot from 2017-11-09 10-30-23]<https://user-images.githubusercontent.com/9088470/32613654-7d38f634-c539-11e7-8d62-03fc1725714d.png>

aparrapo, 2017-11-10:
Hi @Tangelo1. How did you install opencv? It seems ${OpenCV_LIBS} cannot find the libraries.

Tangelo1, 2017-11-16:
Well I originally built it manually, then I decided to use this script https://github.com/milq/milq/blob/master/scripts/bash/install-opencv.sh

closed 2017-11-28 by Tangelo1