facebookarchive/Surround360 #255: Isn it should be I0.rows at errorFunction?

bona00ku, 2018-03-30:
At surround360_render/source/optical_flow/PixFlow.h,
I found a mysterious code.

At line 520, when calculating horizontal RegularizationCoef related error, 
shouldn't it be 
                         `horizontalRegularizationCoef * fabsf(flowDir.x) / float(I0.rows);` 
instead of 
                        ` horizontalRegularizationCoef * fabsf(flowDir.x) / float(I0.cols);` 
?

aparrapo, 2018-03-30:
Nice catch @bona00ku! Please feel free to create a pull request with the fix so we can give you proper recognition :)

aparrapo, 2018-04-15:
@rumblecoder you are totally right, I read it wrong the first time. Please go ahead and create a pull request. Thanks!

rumblecoder, 2018-04-16:
Although it makes sense to use I0.rows for vertical regularization and I0.cols for horizontal regularization, my suggestion would conflict with [this](https://github.com/facebook/Surround360/pull/148) previous rejected pull request.
Using I0.cols on both terms was on purpose (a comment in the code here would be helpful). So just reverting the last commit should be fine.