facebookarchive/Surround360 #265: Unable to run run_all.py on the sample dataset http://surround360.hacktv.xyz/sample/sample_dataset.zip

rinka-meltiron, 2018-07-04:
This also references #66 
I have two issues related to this:
1. The image sample_result/gui_config.png does not match the UI of run_all.py and so I tried to wing it.  Please will you update this png with every release, it will help us all.
2. Here's the data I filled out in run_all.py.

[image]
```
-------- Wed Jul 04 2018 16:06:37 IST -------

python /home/rinka/Documents/fb-git/Surround360/surround360_render/scripts/run_all.py --data_dir /home/rinka/Desktop/in/sample_dataset/vid/000000/raw --start_frame 1 --quality 6k --cubemap_height 0 --steps_render  --verbose  --dest_dir /home/rinka/Desktop/out --frame_count 0 --cubemap_format photo --cubemap_width 0

Checking required files...
WARNING: Calibration file not found. Using default file.

WARNING: Pole masks not found. Using default files.

ERROR: No RGB images in /home/rinka/Desktop/out/rgb/cam0
```
and on checking the directory ./out here's what I get:
```
[Wed Jul  4 16:07:52 IST 2018]
rinka@rinka-GPU-2-Desktop:~/Desktop/in
$ ls ../out
config  logs  pole_masks  runtimes.txt
```
Repeating the run with cubemap_format set to video, here's what I get:
```
-------- Wed Jul 04 2018 16:06:37 IST -------

python /home/rinka/Documents/fb-git/Surround360/surround360_render/scripts/run_all.py --data_dir /home/rinka/Desktop/in/sample_dataset/vid/000000/raw --start_frame 1 --quality 6k --cubemap_height 0 --steps_render  --verbose  --dest_dir /home/rinka/Desktop/out --frame_count 0 --cubemap_format photo --cubemap_width 0

Checking required files...
WARNING: Calibration file not found. Using default file.

WARNING: Pole masks not found. Using default files.

ERROR: No RGB images in /home/rinka/Desktop/out/rgb/cam0

-------- Wed Jul 04 2018 16:10:56 IST -------

python /home/rinka/Documents/fb-git/Surround360/surround360_render/scripts/run_all.py --data_dir /home/rinka/Desktop/in/sample_dataset/vid/000000/raw --start_frame 1 --quality 6k --cubemap_height 0 --steps_render  --verbose  --dest_dir /home/rinka/Desktop/out --frame_count 0 --cubemap_format photo --cubemap_width 0

Checking required files...
ERROR: No RGB images in /home/rinka/Desktop/out/rgb/cam0

-------- Wed Jul 04 2018 16:11:07 IST -------

python /home/rinka/Documents/fb-git/Surround360/surround360_render/scripts/run_all.py --data_dir /home/rinka/Desktop/in/sample_dataset/vid/000000/raw --start_frame 1 --quality 6k --cubemap_height 0 --steps_render  --verbose  --dest_dir /home/rinka/Desktop/out --frame_count 0 --cubemap_format video --cubemap_width 0

Checking required files...
ERROR: No RGB images in /home/rinka/Desktop/out/rgb/cam0
```
Essentially the same issue.
Please will you make the NOTES.txt more elaborate.  For example: I'm not clear what you mean by:
```
This directory structure reflects the state after the "unpack", "arrange", and "isp" steps of the pipeline have already been run. This means you do not need to run these steps again. We also provide the RAW files which are the inputs to the ISP, which you may use as input to run the "isp" step of the pipeline as well.
```
While unpack button is available on run_all.py, I am not clear how the "arrange" and the "isp" steps were done.  Also, I don't see how I can refer to the files created by these steps (important when I start creating my own videos and cameras).

I have issues with Calibration.md too but I'll create a separate issue around that.

Dancewine, 2018-07-17:
I have the same question...Have you solved this problem?

McDifference, 2018-08-09:
I have the same problem when I try to render the sample dataset. 
ERROR: No RGB images in /sample_dataset/rgb/cam0
What should be in /rgb/cam0?

BigGordon, 2018-08-24:
I solve this problem by 
1、creating a directory named rgb in the destination directory, and also create cam0~cam16 directorys in rgb. 
2、Then I copied png from sample dateset and name it like /rgb/cam0/000000.png~/rgb/cam16/000000.png for the fist frame. 
And it will work. It's a heavy solution but I don't know other better ways.

McDifference, 2018-08-24:
Thank you @BigGordon , I have solved the problem by doing the same thing you did.

rinka-meltiron, 2018-08-25:
Yes this worked for me too.  I'm closing the issue.

closed 2018-08-25 by rinka-meltiron