facebookarchive/Surround360 #270: Issue with folly when building surround360_render

bluelikemilk, 2019-02-11:
Hi,

I have already fixed the problem on folly && double-conversion #263 
But then I got another error **undefined reference to `cplus_demangle_v3_callback'**
I have rebuilt folly several times but nothing changes. Is there any way to fix this problem?

[ 45%] Linking CXX executable bin/TestVignettingCalibration
//usr/local/lib/libfolly.a(Demangle.cpp.o): In function `folly::detail::cplus_demangle_v3_callback_wrapper(char const*, void (*)(char const*, unsigned long, void*), void*)':
/home/chufan/folly/folly/detail/Demangle.cpp:38: undefined reference to `cplus_demangle_v3_callback'
collect2: error: ld returned 1 exit status
CMakeFiles/TestVignettingCalibration.dir/build.make:133: recipe for target 'bin/TestVignettingCalibration' failed
make[2]: *** [bin/TestVignettingCalibration] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/TestVignettingCalibration.dir/all' failed
make[1]: *** [CMakeFiles/TestVignettingCalibration.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Thanks for your help!

Dorothy-2016, 2019-02-21:
Hi, I encountered the same question about folly linking in surround360_render as you like follows:

undefined reference to `folly::f14::detail::F14LinkCheck<(folly::f14::detail::F14IntrinsicsMode)2>::check()'

Have you foud out any solution?

bluelikemilk, 2019-02-21:
Not yet. I searched in google but it seems that nobody posted this issue
before.

RoshanPAN, 2019-10-06:
Also met this issue when trying to use `folly/Random.h`

niloup, 2019-11-15:
I am encountering similar issues with folly:

lib/libLibUtil.a(RawUtil.cpp.o): In function `folly::f14::detail::F14Table<folly::f14::detail::NodeContainerPolicy<folly::dynamic, folly::dynamic, folly::detail::DynamicHasher, folly::detail::DynamicKeyEqual, void> >::rehashImpl(unsigned long, unsigned long, unsigned long, unsigned long, unsigned long)::{lambda()#1}::operator()() const':
RawUtil.cpp:(.text._ZZN5folly3f146detail8F14TableINS1_19NodeContainerPolicyINS_7dynamicES4_NS_6detail13DynamicHasherENS5_15DynamicKeyEqualEvEEE10rehashImplEmmmmmENKUlvE_clEv[_ZZN5folly3f146detail8F14TableINS1_19NodeContainerPolicyINS_7dynamicES4_NS_6detail13DynamicHasherENS5_15DynamicKeyEqualEvEEE10rehashImplEmmmmmENKUlvE_clEv]+0x75): undefined reference to `folly::f14::detail::F14LinkCheck<(folly::f14::detail::F14IntrinsicsMode)2>::check()'
collect2: error: ld returned 1 exit status
CMakeFiles/ViewForegroundMaskThresholds.dir/build.make:159: recipe for target 'bin/ViewForegroundMaskThresholds' failed
make[2]: *** [bin/ViewForegroundMaskThresholds] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ViewForegroundMaskThresholds.dir/all' failed
make[1]: *** [CMakeFiles/ViewForegroundMaskThresholds.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I am using Ubuntu 16.04. 

@aparrapo : Any help is greatly appreciated. Thank you!

aparrapo, 2019-12-01:
The version of folly that has consistently worked for us is v2018.08.06.00. Try that one and see if the problem goes away:
```
git clone https://github.com/facebook/folly.git
cd folly && git checkout v2018.08.06.00
mkdir build && cd build
cmake configure .. -DBUILD_SHARED_LIBS:BOOL=ON -DCMAKE_POSITION_INDEPENDENT_CODE:BOOL=ON
make -j && sudo make install -j
```